genodelabs/genode #4419: black_hole: provide Event service

m-stein, 2022-02-11:


nfeske, 2022-02-11:
Nice! Merged to staging right away.

m-stein, 2022-02-11:
I added 2 more commits related to the server:
71e14264cf black_hole: add config.xsd
940e998672 black_hole: test run-script

m-stein, 2022-02-11:
Note that I became insecure while trying to add tests for the opened Audio_in/Audio_out sessions. So, I skipped it for now. Maybe someone more experienced can add a short representive access pattern for these two.

nfeske, 2022-02-11:
@m-stein your additions are very much appreciated. I merged both commits to staging.

nfeske, 2022-02-11:
As the test is pretty much free-standing, we may consider turning it into a depot_autopilot test as a subsequent step.

m-stein, 2022-02-12:
I just added the Nic service as well:
a7b82db33e black_hole: provide Nic service

nfeske, 2022-02-14:
> I just added the Nic service as well:

That's very nice!

m-stein, 2022-02-15:
That'll be the last one for this issue:
5d74390e6b black_hole: provide Uplink service

nfeske, 2022-02-15:
Thank you @m-stein. It is very good to see the component becoming more and more complete. I merged your commit to staging.

m-stein, 2022-02-15:
Thanks, I'm happy that you like it :)

nfeske, 2022-02-17:
The depot_autopilot reported the following problem on OKL4, seL4, and L4/Fiasco:

```
[init -> depot_autopilot] 2.577 [init -> test-black_hole] Error: Uncaught exception of type 'Test::Main::_submit_uplink_pkts()::Uplink_packet_alloc_failed'
[init -> depot_autopilot] 2.577 [init -> test-black_hole] Warning: abort called - thread: ep
```

m-stein, 2022-02-18:
@nfeske Thanks for the hint!
This fixes the problem for me on okl4/fiasco/sel4 and also runs fine on nova+x86_64 (my initial testing platform):
eb23bad352 test/black_hole: fix failing packet allocation

nfeske, 2022-02-19:
Thank you attending the problem. I cherry-picked your commit to staging.

When looking at the changes, the repeated use of prefixed `_uplink` members caught my attention. As a matter of style, have you considered grouping such closely related members in a dedicated `struct`? The naming of the members of such a local struct would probably become more concise, hopefully defeating the tendency of the ever growing characters per line when formatting the code as one single tabular style. That's just a thought, I'm not urging you. :-)

m-stein, 2022-02-20:
I considered it indeed, but refrained from doing so because I sensed it still manageable with the given complexity. However, here we go ^_^
49a1b3f69c test/black_hole: test link-state change at uplink
088220d974 test/black_hole: split-up test classes

m-stein, 2022-02-20:
I went a bit further and created classes with dedicated interfaces instead of local structs. I hope this is fine as well?

nfeske, 2022-02-21:
> I hope this is fine as well?

That's even much better! It is great that you took my little remark so far.

I merged the two new commits to staging.

Since the original test/black_hole commit has not reached master, I wonder if we should better squash the commits together? Or do you wish to preserve the intermediate states of the test/black_hole component in the history?

m-stein, 2022-02-21:
@nfeske Not at all, you may squash all commits regarding the test together. Thanks for asking :)

nfeske, 2022-03-10:
The black-hole test fails on some nightly tested platforms such as qemu/nova/x86_32 due to an uncaught `Packet_alloc_failed` exception in the uplink test. I tried to address the failures with 47fd1c5cb3f6999435d24d5adbd5ff68862ede99 but this change is apparently not correct because it triggers different symptoms (e.g., the test gets stuck on qemu/pistachio/x86_32). I'd prefer to leave the fix to @m-stein and reverted my presumed fix from staging now.

m-stein, 2022-03-11:
I'd do this. I think 22.05 is realistic even though I'm occupied with Wireguard.

nfeske, 2022-03-11:
Thanks @m-stein for responding. As the upcoming Sculpt is scheduled for end of March, I'm aiming at a much sooner fix though.

m-stein, 2022-03-11:
Oh, sorry, I misread your message and thought of the Genode release. I could take care of the ROM service till end of March though.

m-stein, 2022-03-12:
Done:
e688367575 black_hole: provide ROM service

nfeske, 2022-03-14:
Thank you Martin for the super quick response!

May I request two minor changes?

* The combination of `String` with `memcpy` should better be replaced by a single call of `copy_cstring`, which can be found in util/string.h
* I'd encourage using the `Session_object` class located at base/session_object.h for all RPC objects that implement session interfaces. Even though your ROM-session implementation does not benefit from it (as it is so simple), it would still be good to foster consistency towards future-proof patterns. This way, someone who takes the black-hole component as blue print for a custom ROM server would copy the best pattern.

BTW, after having slept over the need for the GPU and USB services again, I came to the realization that it would suffice to merely announce those services and respond to session requests with `Service_denied`. So we could get away with just implementing the `Root::Component` parts of those services.

m-stein, 2022-03-14:
Oops, I had already implemented the Gpu service in a way that one can construct a Gpu::Connection with black_hole as server when I read your last comment:

48c14fde57 black_hole: provide Gpu service

I will soon apply your two comments to the ROM commit and take care of the test failures you observed.
Shall I revert the GPU commit in favour of the Service_denied solution?

m-stein, 2022-03-14:
In my opinion, a service that simply gets denied is not so much in line with the rest of the black_hole server, but I can see that it would be opportune in the Sculpt context.

nfeske, 2022-03-14:
> Shall I revert the GPU commit in favour of the Service_denied solution?

Either way is fine by me. My suggestion was merely aiming at minimizing the effort needed to barely satisfy the needs of Sculpt.

m-stein, 2022-04-07:
Here's for USB:
fed4546526 quickfix "black_hole: provide Usb service"

chelmuth, 2022-04-08:
I merged 4f70e5d1b50012432a149eed5746a2405b4e885b to staging.

@m-stein are you aware of recurring nightly fails of test-black_hole? Any chance to get them addressed soon?

```
[2022-04-07 03:00:36] [init -> depot_autopilot] --- Run "test-black_hole" (max 10 sec) ---
[2022-04-07 03:00:36] [init -> depot_autopilot] 
[2022-04-07 03:00:38] [init -> depot_autopilot] 1.801 [init -> test-black_hole] Error: Uncaught exception of type 'Black_hole_test::Uplink_test::_submit_pkts()::Packet_alloc_failed'
[2022-04-07 03:00:38] [init -> depot_autopilot] 1.820 [init -> test-black_hole] Warning: abort called - thread: ep
[2022-04-07 03:00:46] [init -> depot_autopilot] 
[2022-04-07 03:00:46] [init -> depot_autopilot]  test-black_hole                 failed     9.983  timeout 10 sec
```

m-stein, 2022-04-08:
Yes, I'm at it.

m-stein, 2022-04-08:
@chelmuth Sorry for the delay:
25eb4bc183 test/black_hole: fix "packet alloc failed"
c38d5e9c91 black_hole: serve real USB sessions

nfeske, 2022-04-13:
Thanks @m-stein. The commit series entered the master branch.

commit e731b620 (2022-02-11): black_hole: provide Event service

Fixes #4419
closed 2022-04-13 by nfeske