genodelabs/genode #4420: Drop `_drv` suffix from device driver executables

nfeske, 2022-02-11:
The suffix was introduces as an ad-hoc nomenclature when we started Genode in 2006. But when reflecting on it, it served no real purpose. I'd like to eventually drop it (not today but over the next months) for two reasons:

* The suffix violates our established style of naming things without abbreviations. (https://genode.org/documentation/developer-resources/conventions) The `_drv` suffix is one of the last remaining offenders.

* It is ambiguous. It suggests a classification of components where no clear line could be drawn. E.g., not all components hosted in the `src/drivers/` directories are suffixed that way anyway. E.g., timer drivers are named as `timer`, or `fb_sdl` is driver-like but does not carry the suffix. Other components like `dummy_rtc_drv` carry the suffix without being a driver.

Given these observations, I suggest dropping this little bureaucratic relic. I wonder, would there by any objections?

nfeske, 2023-02-21:
While we are at it, @chelmuth recently pointed out that the name of the `<repo>/src/drivers/` directory is inconsistent with the sibling directories `test/`, `app/`, `server/`. It is the only occurrence of plural. Should we take this issue as opportunity to rectify this?

nfeske, 2024-05-31:
The commits named "Remove '_drv' suffix" implement the first of the two changes. The renaming of src/drivers to src/driver will follow.

nfeske, 2024-06-03:
Just merged the second part of this issue (the renaming of src/drivers to src/driver) to the respective staging branches.

nfeske, 2024-06-03:
@ssumpf thanks for the hint, fixup 56d491d63307c3e6e18aa9dc6d998cd54db75e7d corrects this glitch.

ssumpf, 2024-06-03:
@nfekse: 0d7d6a6 contains a missing piece in Mesa.

ssumpf, 2024-06-03:
_gems/sculpt/deploy/goa_testbed_, _gems/run/mixer_gui_qt_test.run_, and _gems/run/tiled_wm.run_ still have references to `mesa_gpu_drv.lib.so`

nfeske, 2024-06-04:
Thanks @ssumpf.

nfeske, 2024-06-20:
Fixed in master.

closed 2024-06-20 by nfeske