genodelabs/genode #4426: make bender UEFI aware

trimpim, 2022-02-14:
In a recent discussion on the genode mailing list [1] alex-ab asked me to create an issue about making bender aware of UEFI.

The current bender placed a wrong base address for the first serial port in to the BDA, and therefore I didn't receive any log output.

[1] https://lists.genode.org/pipermail/users/2022-February/007935.html

trimpim, 2022-04-14:
@alex-ab unfortunately on my particular hardware doesn't help.

I have tried to boot the following:
 - `image/uefi` via ipxe  ( to enable this I modified `run/load/ipxe` slightly )
 - `image/disk` from USB stick

trimpim, 2022-06-02:
@alex-ab on my i7 11700 test system, which uses a `MSI Edge Gaming WiFi` mainboard, your patch helps with the serial output.

alex-ab, 2022-06-06:
@trimpim: thank you for testing the commit and reporting. Sorry for the delay on my side, I was bit distracted with other topics. I added another debug commit, which I would ask you, please, to try on the machine which did not work. The commit mainly sets the uart to 0x3f8 (which is the working one for the problematic machine!?) and adds debug printfs which should help me/you to understand why it does not work for you.

trimpim, 2022-06-13:
@alex-ab Sorry for the delay. I was working from home office where I do not have the correct hardware to test.

When I cherry-pick your commit an run (log/run) on the problematic machine I get the following output:

```
 Looking for serial controllers on the PCI bus...
  none found.
A iobase=0 efi_boot=0 serial_fallback=1
B iobase=21c efi_boot=0 serial_fallback=1
C iobase=21c efi_boot=0 serial_fallback=1
```

I guess, this is what you were after. Otherwise let me know so I can provide more accurate information.

alex-ab, 2022-06-14:
According to your output, the system seems to be in non UEFI mode and bender finds in the BDA a configured serial adapter, so the serial_fallback option is not considered. As it looks it works as it was designated, but still it does not help you ... I guess, an option to enforce your i/o ports is what you want ?

trimpim, 2022-06-14:
@alex-ab I will check if I can set the system to UEFI mode (I wasn't aware of this fact) and if this helps with the log out put.

I will keep you informed on my findings, and if further changes are of needed.

trimpim, 2022-06-17:
As far as I can tell, the problematic device [1] only boots UEFI images. I have a rather ugly workaround to get the serial connection to work [2].

[1] UPX-EDGEI7-A10-1664-F01
[2] https://github.com/trimpim/genode/tree/up_xtreme-bender_patch  /  https://github.com/trimpim/morbo/tree/up_xtreme

chelmuth, 2022-07-07:
I'd like to report that I managed to fix bender to successfully detect UEFI on UPX-EDGEI7-A10-1664-F01 with https://github.com/chelmuth/morbo/commit/1534b1705734f1626a600ecfa24d294b7f0a68e7. Thanks to @trimpim for his hints to boot Genode on the UPX easily.

@alex-ab feel free to merge my commit with your adaptions and update bender in Genode.

alex-ab, 2022-07-11:
@chelmuth, @trimpim: Thanks for your efforts and finally a working version :-). I added the updated bender and also the grub2 update(avoiding potential memory corruption on UEFI and BDA region).

trimpim, 2022-07-20:
@chelmuth sorry for coming in late.

We tried the latest staging on the UP-Xtreme hardware, but don't get any log output with `run/log`.  We also tried resetting the BIOS to default values. 

This made me think what could be the difference between your board and ours?
  - BIOS version (our is 2.20.1275)
  - did you change any specific settings for it to work?

(FYI I will be out of office from 22.07. to 02.08.)

chelmuth, 2022-07-20:
> We tried the latest staging on the UP-Xtreme hardware, but don't get any log output with `run/log`. We also tried resetting the BIOS to default values.
> 
> This made me think what could be the difference between your board and ours?
> 
>     * BIOS version (our is 2.20.1275)

Same here.

>     * did you change any specific settings for it to work?

Not that I'm aware off. I use

- Ethernet port labeled "LAN1"
- USB stick prepared by iPXE (commit d3c8944d5c3ab26282) with `make bin-x86_64-efi/ipxe.usb`
- COM port labeled "COM1"
- Reset via AMT
- Genode genodelabs/genode@ec48368d01ac861910c26c9fd4e28bd6a77300fe

and the following RUN_OPTS.

```
RUN_OPT  = --amt-tool wsman
RUN_OPT += --include power_on/amt
RUN_OPT +=   --power-on-amt-host upx1-amt.genode.labs
RUN_OPT +=   --power-on-amt-password '<password>'
RUN_OPT +=   --power-on-amt-timeout 14
RUN_OPT += --include image/uefi
RUN_OPT += --include load/ipxe
RUN_OPT +=   --load-ipxe-base-dir /tftpboot/devel
RUN_OPT +=   --load-ipxe-boot-dir /upx1
RUN_OPT += --include log/serial
RUN_OPT +=   --log-serial-cmd 'picocom -e p -b 115200 /dev/ttyUSB0'
```

> (FYI I will be out of office from 22.07. to 02.08.)

I'm also away July 22 to August 8.

chelmuth, 2022-07-21:
@trimpim I uploaded my working example [log.zip](https://github.com/genodelabs/genode/files/9156638/log.zip) for testing on your device.

chelmuth, 2022-08-09:
To wrap up the open question: @trimpim and me checked the UPX settings and discovered AMT SOL redirection was enabled for some unknown reason. Bender's original purpose is to scan the PCI bus for serial devices and update the BIOS data area (BDA) if a device was found. Afterwards, NOVA and Genode core take this information for serial log settings. As AMT provides a virtual PCI serial device if SOL is enabled, bender did its job on the device and all log messages disappeared into nirvana.

closed 2022-08-17 by chelmuth