genodelabs/genode #4427: vfs/lwip: add missing signal handlers

jschlatow, 2022-02-14:
The vfs_lwip plugin does not handle ack_avail nor ready_to_submit signals from the tx channel. In scenarios like netperf in which data flows into a single direction, it may occur that the sender waits for an ack but the receiver (using vfs_lwip) was not able to send it immediately because the TX packet queue was full. With the missing signal handling, however, the receiver will only wake up when it receives another packet from the RX queue.

jschlatow, 2022-02-14:
Commit 0b820ed solves the low-throughput problem that I observed when running netperf server on Sculpt and netperf client in a Linux VM on Sculpt. Without the fix, I merely achieved a throughput of 30Mbit/s. With the fix, I get over 80Mbit/s on a 100Mbit/s network.

On the downside, I also observed a throughput degradation in other scenarios with this fix because it assumes the Nic server to apply batching. When I remove one of the tx signal handlers, the throughput remains the same, yet this will be incompatible with a batching Nic server.

For the record, I'm using the following netperf command on the client side to output some statistics:
```
netperf-2.6.0 -4 -j -H <server-ip-address> -t tcp_stream  -- -P 18000 -k THROUGHPUT,THROUGHPUT_UNITS,LOCAL_TRANSPORT_RETRANS,MAX_LATENCY,MIN_LATENCY,MEAN_LATENCY
```

The missing signal handling increases the MAX_LATENCY and MEAN_LATENCY significantly, which explains the bad throughput. However, even with the fix, I still see about 600 retransmissions iff the client runs in a VM on Sculpt, which I cannot explain yet.

nfeske, 2022-02-14:
Thank you for the fix and the very interesting findings.

Down the road, we may consider consolidating the distinct RX and TX signal handlers into only one signal handler, following a similar like of reasoning as in #4388.

> On the downside, I also observed a throughput degradation in other scenarios with this fix because it assumes the Nic server to apply batching.

Fortunately, that can be addressed by implementing the batching in only one place, the NIC router, right?

jschlatow, 2022-02-14:
> Fortunately, that can be addressed by implementing the batching in only one place, the NIC router, right?

Correct. Though it will also be beneficial to implement batching in the drivers, i.e. the Uplink_client_base, as well.

nfeske, 2022-02-15:
Aye aye. Merged to staging.

nfeske, 2022-02-15:
Fixed in master.

closed 2022-02-15 by nfeske