genodelabs/genode #4430: trace_buffer: fix several corner cases

jschlatow, 2022-02-17:
By implementing a test component for the trace buffer, I exposed a few corner cases that need fixing.

jschlatow, 2022-02-17:
Commit 181900b adds the test pkg for depot_autopilot. It currently skips testing a slow consumer with a fast producer since this is known not to work.

Commit 3c8c37f fixes an out-of-bounds read when iterating the buffer entries.

Commit cf1000b fixes an unlikely corner case when the iteration is stopped right after the buffer wrapped.

Note, that I still plan to revise the trace buffer implementation and probably fix all the issues more elegantly. I'll open another issue for this though.

jschlatow, 2022-02-17:
Removed the fixed label because I found a few more issues when running the test on a native Linux instead of a VM.

nfeske, 2022-02-18:
Thank you @jschlatow for the commit series and the detailed descriptions. It seems that the critical inspection of the trace utilities was long overdue. I very much appreciate this work, and merged all the commits to staging.

nfeske, 2022-02-28:
The commits entered the master branch.