genodelabs/genode #4431: Xml_node: support blackslash as attribute value

nfeske, 2022-02-17:
XML allows attribute values like `<node attr="\"/>`. The current version of the XML parser wrongly reflects this case as `Invalid_syntax`.

m-stein, 2022-02-17:
Commit add2011ff1 seems to break building the vfs lib:
```
~/genodelabs/genode/build/x86_64$ LIB=vfs make 
checking library dependencies...
  Library ldso_so_support
  Library ld
  Library base
  Library vfs
    COMPILE  file_system_factory.o
In file included from /home/lypo/genodelabs/genode/repos/base/include/util/xml_node.h:18,
                 from /home/lypo/genodelabs/genode/repos/os/include/vfs/types.h:19,
                 from /home/lypo/genodelabs/genode/repos/os/include/vfs/directory_service.h:17,
                 from /home/lypo/genodelabs/genode/repos/os/include/vfs/file_system.h:17,
                 from /home/lypo/genodelabs/genode/repos/os/include/vfs/env.h:17,
                 from /home/lypo/genodelabs/genode/repos/os/include/vfs/file_system_factory.h:17,
                 from /home/lypo/genodelabs/genode/repos/os/src/lib/vfs/file_system_factory.cc:16:
/home/lypo/genodelabs/genode/repos/base/include/util/token.h: In instantiation of ‘Genode::size_t Genode::Token< <template-parameter-1-1> >::_quoted_string_len(Genode::size_t) const [with SCANNER_POLICY = Vfs::Tar_file_system::Scanner_policy_path_element; Genode::size_t = long unsigned int]’:
/home/lypo/genodelabs/genode/repos/base/include/util/token.h:249:12:   required from ‘Genode::size_t Genode::Token< <template-parameter-1-1> >::_calc_len(Genode::size_t) const [with SCANNER_POLICY = Vfs::Tar_file_system::Scanner_policy_path_element; Genode::size_t = long unsigned int]’
/home/lypo/genodelabs/genode/repos/base/include/util/token.h:88:44:   required from ‘Genode::Token< <template-parameter-1-1> >::Token(const char*, Genode::size_t) [with SCANNER_POLICY = Vfs::Tar_file_system::Scanner_policy_path_element; Genode::size_t = long unsigned int]’
/home/lypo/genodelabs/genode/repos/os/src/lib/vfs/tar_file_system.h:316:43:   required from here
/home/lypo/genodelabs/genode/repos/base/include/util/token.h:200:59: error: ‘end_of_quote’ is not a member of ‘Vfs::Tar_file_system::Scanner_policy_path_element’
  200 |    for (; i + 1 < max_len && !SCANNER_POLICY::end_of_quote(&_start[i]); i++)
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [/home/lypo/genodelabs/genode/repos/base/mk/generic.mk:62: file_system_factory.o] Error 1
make[1]: *** [var/libdeps:80: vfs.lib] Error 2
make: *** [Makefile:312: gen_deps_and_build_targets] Error 2
```

nfeske, 2022-02-17:
Thanks @m-stein for reporting. Fixup 14a6f2f9de1dead7d6edd078531f9d6edc9869c9 addresses this issue.

nfeske, 2022-02-17:
Fixup c0103747da97c9e232f4b0ada6d6430a5d3acffc covers another instance (libc).

m-stein, 2022-02-19:
@nfeske Thanks for the having fixed it so fast!

commit add2011f (2022-02-17): xml_node: support backslash as attribute value

XML allows attribute values like <node attr="\"/>. The XML parser
wrongly reflects this case as 'Invalid_syntax'. This behavior stems from
the implicit use of the 'end_of_quote' function, which considers the
sequence of '\"' as a quoted '"' rather than the end of a quoted string.

The patch solves this problem by making the 'end_of_quote' part of
the tokenizer's scanner policy.

The patch removes the 'end_of_quote' function from 'util/string.h'
because it is not universal, and to avoid the ambiguity with
'SCANNER_POLICY::end_of_quote'.

Fixes #44 [... cut]
commit c2efa540 (2022-02-21): xml_node: support backslash as attribute value

XML allows attribute values like <node attr="\"/>. The XML parser
wrongly reflects this case as 'Invalid_syntax'. This behavior stems from
the implicit use of the 'end_of_quote' function, which considers the
sequence of '\"' as a quoted '"' rather than the end of a quoted string.

The patch solves this problem by making the 'end_of_quote' part of
the tokenizer's scanner policy.

The patch removes the 'end_of_quote' function from 'util/string.h'
because it is not universal, and to avoid the ambiguity with
'SCANNER_POLICY::end_of_quote'.

Fixes #44 [... cut]
closed 2022-02-28 by chelmuth