genodelabs/genode #4433: rump: _wait_and_dispatch_one_io_signal called from non-entrypoint thread "pdaemon"

cproc, 2022-02-21:
When I run virtualbox6 with rump on the current staging branch, the following warning appears and the (Debian) VM boot process does not reach the GUI.

```
[init -> rump_fs] Warning: _wait_and_dispatch_one_io_signal called from non-entrypoint thread "pdaemon"
```

nfeske, 2022-02-21:
I suspect that this problem is related to the change 911ff3170927525bad921720945f3a361661c2bd, which replaced the former use of blocking packet-stream functions with an explicit `wait_and_dispatch_one_io_signal` loop. During my tests, all calls of the `submit` function were driven by the entrypoint thread. But I was probably too optimistic.

Unfortunately, I won't be able to take a deeper look because I'm offline for the remainder of the week. @cproc could you take care of the investigation?

Let me share a rough plan for cases like this where a blocking operation for an I/O signal might be issued from a thread different from the entrypoint that is associated with an I/O signal handler. (BTW, this pattern is needed at a few places such as the timer connection's `msleep` function, or `Env::_block_for_session)

Instead of calling `ep.wait_and_dispatch_one_io_signal()` in cases like these, I'd introduce a wrapper object that takes the role of an I/O signal handler and offers a method for blocking, like sketched below:

```
class Io_signal_blockade : Signal_handler<Io_signal_blockade>
{
  private:

    void _handle_io_signal();

    Thread const * _ep_thread_ptr;

    unsigned _cnt; /* counts calls of _handle_io_signal */

    Semaphore _sem;

  public:

    Io_signal_blockade(Entrypoint &ep);

    void block_for_io();
};
```

Once instantiated, e.g., in the vfs plugin, it could be passed as I/O signal handler to a block session. The `block_for_io` method allows for blocking for this specific signal. Internally, this method distinguishes two cases:

* If called from the entrypoint thread, according the `Thread::myself`, it calls `ep.wait_and_dispatch_one_io_signal` until the signal-handler method `_handle_io_signal` was executed. This could be detected by tracking a counter value that is incremented each time `_handle_io_signal` gets triggered.
* If called from a different thread, it would block on a semaphore, which is released by the `_handle_io_signal` handler.

That's of course just a rough sketch. The interesting part of the implementation of `_handle_io_signal`, driving both `_cnt` and `_sem`. I think such a utitlity would not only solve the rump problem at hand but become a very useful addition to `base/signal.h` that we can then also use in the other situations mentioned above.

@cproc Would you give implementing this idea a try?

You will probably stumble upon the detail of how to get hold of the `Thread *` of an entrypoint. One way to do this would be triggering the signal once via `ep.local_submit` at construction time. Then, from within the `_handle_io_signal` method (that is known to be executed in the context of the ep thread), you may capture the pointer returned by `Thread::myself` and store it in `Io_signal_blockade::_ep_thread_ptr`. As a last resort - should this idea not work out - we could add testing method to the entryoint, like `bool Entrypoint::belongs_to_thread(Thread const &)`.

cproc, 2022-02-24:
Commit 65947ed is a first attempt to fix the problem. I experimented with the suggested semaphore and counter for a while, but then it appeared simpler to use a registry of blockades instead. Unfortunately, the 'local_submit' solution to get hold of the ep thread pointer did not work, because the signal handler did not get called before the first call of 'block_for_io'. With this first version the pointer is derived from the thread which calls the 'Io_signal_blockade' constructor and the class is still located in the rump backend, but it seems to solve the problem at hand at least and might be sufficient for the release.

nfeske, 2022-02-28:
The commit entered the master branch.

closed 2022-02-28 by nfeske