genodelabs/genode #4434: trace_buffer: eliminate potential corruption of consumer state by the producer

jschlatow, 2022-02-21:
As already noted in #4244 and #4430, the current trace buffer implementation does not quite handle the case of a slow consumer. Despite keeping the count of buffer wraparounds, there is no guarantee that the producer cannot overwrite entries that are currently processed by the consumer. Especially when entries have variable size, this can corrupt the consumer state.

jschlatow, 2022-02-21:
I split the modifications into two commits. Commit d530411 comprises mostly syntactial changes in order to simplify `for_each_new_entry()`.
The second commit, 4ca95d5, introduces a applies a partitioning scheme to prevent overwriting and corruption in case the consumer is too slow. It reuses the old implementation and provides the same interface. In addition, the partitioned buffer is able to keep track of how many entries have been overwritten before being consumed (and therefore lost).

The changes are intended to be merged  after the upcoming release.

jschlatow, 2022-02-21:
Note, that the changes currently break `test-trace_logger`. I'm going to look into that.

jschlatow, 2022-02-22:
> Note, that the changes currently break `test-trace_logger`. I'm going to look into that.

My bad. Working with test-trace_buffer, I completely forgot paying attention to relative addressing to allow passing the buffer between address spaces. Fixed by 01ebe44. Note, that I also stumbled upon #4435 when looking into this.

Another issue that popped up is that the trace logger started accessing the buffer before it was initialised by the traced thread. I think the best solution here is to eagerly initialise the buffer right when it is allocated to prevent any coordination protocol on the TRACE client side (see 5482bc6).

nfeske, 2022-03-09:
Thanks for this feast for the eyes! The documentation added to trace/buffer.h is awesome.

Admittedly, I merged the rework changes without having internalizing the algorithm in detail and I remain a bit reluctant to fully embrace the 50/50 partitioning. The latter remark is just a gut feeling that the split might be wasteful - but on the other hand, conserving memory is a less important concern than the assurance of integrity in usual tracing scenarios.

The added tests and the fixed corner cases give a lot of confidence in the new version. Thanks a lot, @jschlatow!

jschlatow, 2022-03-16:
I'm happy to hear that. Thank you for the feedback!

nfeske, 2022-04-13:
Fixed in master.

closed 2022-04-13 by nfeske