genodelabs/genode #4450: update intel display driver for pc

alex-ab, 2022-03-18:


alex-ab, 2022-03-18:
I pushed a first version of the update as issue_4450 based on staging. As discussed offline, the update is functional and usable with Sculpt, but I still have to check some machines and configuration updates. Nevertheless, all is packaged and should(tm) build and work.

nfeske, 2022-03-21:
Thanks for the excellent commit series. The sensible ordering of the changes is very much appreciated and made reviewing them a breeze. I merged your commits with a few (very minor) cosmetic changes to staging. Given @cnuke's remark about the page ref count earlier today, I replaced the commit b2af53b373984f449ec9e08aaeab09c3f628c3c7 by e2dfe9f32a605d9e3c72b9a41c1a8489bb071dcc though, hoping that's right.

My first build test of both the driver and also the depot archive succeeded without problems.

While giving the driver a quick test using the intel_fb.run script on my x250 test machine, the display stays dark. In the log, I can spot a few error messages that may be related to the problem:

```
[ 0] IOMMU:0xffffffff817e5060 FRR:0 FR:0x2 BDF:0:2:0 FI:0x00000000 (0)
...
...
...
[init -> init_dynamic -> intel_fb_drv] pci_bus 0000:00: busn_res: 405fed70R end is updated to 00
[init -> init_dynamic -> intel_fb_drv] 6clocksource: Switched to clocksource dde_counter
[init -> init_dynamic -> intel_fb_drv] proc_create_seq_private called cnt=1 name=ioports
[init -> init_dynamic -> intel_fb_drv] proc_create_seq_private called cnt=2 name=iomem
[init -> init_dynamic -> intel_fb_drv] proc_create_seq_private called cnt=3 name=timer_list
[init -> init_dynamic -> intel_fb_drv] disabling PPGTT to avoid GPU code paths
[init -> init_dynamic -> intel_fb_drv] Error: io_mem caching attributes are not same !!!UNCACHED , requested WRITECOMBINED
[init -> init_dynamic -> intel_fb_drv] sched_set_fifo called, doing nothing
[init -> init_dynamic -> intel_fb_drv] sched_set_fifo called, doing nothing
[init -> init_dynamic -> intel_fb_drv] sched_set_fifo called, doing nothing
[init -> init_dynamic -> intel_fb_drv] Error: could not read I/O port resource 0x3cc
[init -> init_dynamic -> intel_fb_drv] Error: could not write I/O port resource 0x3c2
[init -> init_dynamic -> intel_fb_drv] Error: could not write I/O port resource 0x3c4
[init -> init_dynamic -> intel_fb_drv] Error: could not read I/O port resource 0x3c5
[init -> init_dynamic -> intel_fb_drv] Error: could not write I/O port resource 0x3c5
[init -> init_dynamic -> intel_fb_drv] Error: Cannot delay that long 300 microseconds
[init -> init_dynamic -> intel_fb_drv] i915 0000:00:02.0: [drm] *ERROR* AUX A/DDI A/PHY A: did not complete or timeout within 10ms (status 0xa4130152)
[init -> init_dynamic -> intel_fb_drv] i915_gem_open: called
...
```

During the review of the commits, a few (arguably superficial) suggestions crossed my mind that you may consider as fixups.

* The lx_emul/initcall_order.h header is expected to be identical for all PC drivers because it depends only on the Linux kernel configuration. So it may be sensible to move it to a central place, like pc/src/include/pc/lx_emul/initcall_order.h
* A few `#pragma once` directives slipped in. To maintain consistency, it would be nice to replace them with include guards.
* The intel_fb.run script assembles the config via a number of consecutive `append` operations. Nowadays, we generally try to move the conditional parts to dedicated functions so that the structure of the config is defined at once. This approach helps to keep the inter-mixing of two different languages at minimum. You may take the sculpt.run script as reference, searching for `log_core_start_node`. It would be nice to see this pattern applied in the intel_fb.run script, too.

alex-ab, 2022-03-21:
The 3 top most commits of my staging branch implements the writecombine changes.

nfeske, 2022-03-21:
Thanks for the super quick response, Alex! I updated the staging branch right away.

nfeske, 2022-03-22:
Commit https://github.com/genodelabs/genode/commit/24899005d1e988d998e6fcd4889a650439cc8dad fixes the build errors of the imx drivers.

Commit https://github.com/genodelabs/genode-allwinner/commit/8219edb7bd7399b5e2deb2d8a45b2ecd359cf1b8 fixes a build error in the allwinner nic/emac driver.

Both are related to the addition of the __alloc_pages to the shadowed mm/page_alloc.c in commit 7a2746293731d41616f4e42f96d1b5446af0c9af.

alex-ab, 2022-03-22:
Thanks for taking care, I just wanted to start to fix it :+1:

alex-ab, 2022-03-24:
Another commit is required for X250, 4884cc4c097cf612de078f249eb5be387674bdc1, so the top most 5 commits of issue_4450 are relevant.

cproc, 2022-03-24:
On my x260, with the 5 commits applied I get a distorted image with the intel_fb.run test. Screen resolution is 1366x768 (the screenshot is scaled down a bit).

[image]

cproc, 2022-03-24:
Log output for the x260:

```
[init -> platform_drv] ECAM/MMCONF range 00000000:00000000.0-0000003f:0000001f.7 - addr [00000000f8000000,00000000fc000000)
[init -> platform_drv] Root bridge: 00000000:00000000.0
[init -> platform_drv] 00000000:0000001c.0 - bridge 02:00.0
[init -> platform_drv] 00000000:0000001c.2 - bridge 04:00.0
[init] child "platform_drv" announces service "Platform"
[ 0] IOMMU:0xffffffff817e6060 FRR:0 FR:0x2 BDF:0:2:0 FI:0x00000000 (0)
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:02.0 succeeded
[init -> init_dynamic -> intel_gpu_drv] Reserved beginning 64M of aperture for platform service
[init -> platform_drv] 00000000:00000002.0 adjust IRQ as reported by ACPI: 11 -> 16
[init -> platform_drv] 00000000:00000002.0 uses MSI vector 0x7f, address 0xfee00018
[init -> init_dynamic -> intel_gpu_drv] Map res:0 base:0xf0000000 size:0x1000000 vaddr:0x11000000
[init -> init_dynamic -> intel_gpu_drv] Found: 'HD Graphics 520 (Skylake, Gen9)' gen=9 rev=7 [0x8086:0x1916] (0:2.0)
[init -> init_dynamic -> intel_gpu_drv]   Resource 0 (MEM): base=0xf0000000 size=0x1000000 
[init -> init_dynamic -> intel_gpu_drv]   Resource 2 (MEM): base=0xe0000000 size=0x10000000 prefetchable
[init -> init_dynamic -> intel_gpu_drv]   Resource 4 (I/O): base=0xe000 size=0x40 
[init -> init_dynamic -> intel_gpu_drv] MGGC_0_2_0_PCI
[init -> init_dynamic -> intel_gpu_drv]   Graphics_mode_select:               0x1
[init -> init_dynamic -> intel_gpu_drv]   Gtt_graphics_memory_size:           0x3
[init -> init_dynamic -> intel_gpu_drv]   Versatile_acceleration_mode_enable: 0x0
[init -> init_dynamic -> intel_gpu_drv]   Igd_vga_disable:                    0x0
[init -> init_dynamic -> intel_gpu_drv]   Ggc_lock:                           0x1
[init -> init_dynamic -> intel_gpu_drv] GGTT
[init -> init_dynamic -> intel_gpu_drv]   vaddr:0x11800000 size:0x800000 entries:1048575 used:16384 aperture_size:0x10000000
[init -> init_dynamic -> intel_gpu_drv]   scratch_page:0x106000 (PA)
[init -> init_dynamic -> intel_gpu_drv] Warning: no clock gating
[init -> init_dynamic -> intel_fb_controller] --- Framebuffer controller ---
[init -> init_dynamic -> intel_fb_controller] 
[init -> init_dynamic -> test-framebuffer] --- Test framebuffer ---
[init -> init_dynamic -> intel_fb_drv] --- Intel framebuffer driver started ---
[init -> init_dynamic -> intel_fb_drv] 6NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 16
[init -> init_dynamic -> intel_fb_drv] 6clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[init -> init_dynamic -> intel_fb_drv] 6clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[init -> init_dynamic -> intel_fb_drv] 6PCI: Probing PCI hardware
[init -> init_dynamic -> intel_fb_drv] 6PCI host bridge to bus 0000:00
[init -> init_dynamic -> intel_fb_drv] pci_bus 0000:00: root bus resource 124b380R
[init -> init_dynamic -> intel_fb_drv] pci_bus 0000:00: root bus resource 124b340R
[init -> init_dynamic -> intel_fb_drv] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[init -> init_dynamic -> intel_fb_drv] pci 0000:00:00.0: [8086:1904] type 00 class 0x00060000
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> platform_drv] Warning: 00000000:00000000.0 write access to address=0x30 value=0xfffff800 size=0x2 got dropped
[init -> platform_drv] Warning: 00000000:00000000.0 write access to address=0x30 value=0x0 size=0x2 got dropped
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:00.0 succeeded
[init -> init_dynamic -> intel_fb_drv] pci 0000:00:02.0: [8086:1916] type 00 class 0x00030000
[init -> init_dynamic -> intel_fb_drv] pci 0000:00:02.0: reg 0x10: 18cac0R
[init -> init_dynamic -> intel_fb_drv] pci 0000:00:02.0: reg 0x18: 18cb40R
[init -> init_dynamic -> intel_fb_drv] pci 0000:00:1f.0: [8086:9d48] type 00 class 0x00060100
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:1f.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:1f.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:1f.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:1f.0 succeeded
[init -> platform_drv] intel_fb_drv: assignment of PCI device 00:1f.0 succeeded
[... cut ...]
[init -> init_dynamic -> report_rom]   	<connector name="DP-1" connected="false"/>
[init -> init_dynamic -> report_rom]   	<connector name="HDMI-A-1" connected="false"/>
[init -> init_dynamic -> report_rom]   	<connector name="DP-2" connected="false"/>
[init -> init_dynamic -> report_rom]   	<connector name="HDMI-A-2" connected="false"/>
[init -> init_dynamic -> report_rom]   </connectors>
[init -> init_dynamic -> intel_fb_drv] eDP-1: enable name='1366x768' id=1 1366x768@60
[init -> init_dynamic -> test-framebuffer] green
[init -> init_dynamic -> test-framebuffer] red
```

alex-ab, 2022-03-24:
My X260 here has 1080p and works. I suspect it has to do with the pitch, something similar to 101d052ddb798201a9cf764068a6b7fee55fee8f we will need.

alex-ab, 2022-03-24:
Can you please check fb_info->fix.line_length in register_framebuffer() in repos/pc/src/drivers/framebuffer/intel/pc/fb.c. I suspect that you may get here the expected aligned width size, which then could be used to invoke the lx_emul_framebuffer_ready.

alex-ab, 2022-03-24:
I have to leave for today, in case you internal display stops refreshing on the X260 (as for me) if you disconnect with AMT, you may apply as workaround for the moment in repos/pc/src/drivers/framebuffer/intel/pc/vbt_io_mem.cc just a "return nullptr" at the beginning of the function. With that the internal display works

cproc, 2022-03-25:
> Can you please check fb_info->fix.line_length in register_framebuffer() in repos/pc/src/drivers/framebuffer/intel/pc/fb.c. I suspect that you may get here the expected aligned width size, which then could be used to invoke the lx_emul_framebuffer_ready.

Yes, that fixes the problem (commit 92244d9). I didn't notice a refresh problem when disconnecting AMT so far.

alex-ab, 2022-03-25:
Thanks @cproc for implementing and testing. I added your commit to the issue_4450 branch. Also thanks to the hint of @skalk, I use in the cpu_relax now the udelay_range, which also adjust jiffies implicitly.

chelmuth, 2022-03-25:
I merged commits up to 4efb127f79c20d89b7be94abd81ab784fc278862.

jschlatow, 2022-03-28:
@alex-ab the driver starts up smoothly on my x260 and t490s. Unfortunately, it does not detect the native resolution of my external display, which is 1920x1200. Instead, it only shows resolutions up to 1920x1080 and choses the later.

alex-ab, 2022-03-29:
With the 3 commits on issue_4450 (rebased to staging) the driver works on the new machine too

cnuke, 2022-03-29:
Commit [86a4584](https://github.com/cnuke/genode/commit/86a45846601f013fd76ff042c1835bb2a3bf75d9) makes the `cpu_relax` fix local to the framebuffer driver as the way it is currently implemented (include `linux/jiffies.h`) clash in case of the upcomfing WLAN driver.

alex-ab, 2022-03-29:
> @alex-ab the driver starts up smoothly on my x260 and t490s. Unfortunately, it does not detect the native resolution of my external display, which is 1920x1200. Instead, it only shows resolutions up to 1920x1080 and choses the later.

Thanks for testing. I currently have no explanation why a mode of a connector does not appear. In lx_emul_i915_iterate_modes there are potential "identical" modes skipped, but I can't see atm why this should happen for you.

jschlatow, 2022-03-30:
@alex-ab, I added some debug output to validate that the mode is not skiipped due to some strange reason. It's not listed though.

I tried a few other connectors and combinations. In my default setup, I have the external display connected via display port (more precisely thunderbolt). If I connect it to the HDMI connector of the t490s instead, I get a more interesting picture: The mode is present if I boot with HDMI connected but its missing if I plug HDMI in later on. In the former case, I can unplug/replug the HDMI cable and the mode is always listed.

The same behaviour is reproducible on the x260: The mode only shows up if booted with DP or HDMI connected.

Another, more annoying, issue I discovered is that the internal display of the t490s is frozen no matter if an external display is connected or not. It only seems to refresh once on boot or at the moment when I connect an external display. As soon as the mode of the external display is set, however, the internal display shows a frozen screen.

alex-ab, 2022-03-31:
> Another, more annoying, issue I discovered is that the internal display of the t490s is frozen no matter if an external display is connected or not. It only seems to refresh once on boot or at the [...]

Did you test it with the 4 commits on issue_4450 (not part of staging) or without ?

alex-ab, 2022-03-31:
> The same behaviour is reproducible on the x260: The mode only shows up if booted with DP or HDMI connected.

Yes, I could reproduce it also, but have no explanation.

jschlatow, 2022-03-31:
I tested with staging.

alex-ab, 2022-03-31:
> I tested with staging.

The vbt commit on issue_4450 may help hopefully.

alex-ab, 2022-03-31:
> > > I tested with staging.
> > 
> > 
> > The vbt commit on issue_4450 may help hopefully.
> 
> Indeed. With the additional commits, the internal display works perfectly.

Nice, thanks for testing.

alex-ab, 2022-04-06:
Please consider the topmost 3 commits on issue_4450 for staging. The commit also resolves the issue with the missing  modes of connectors attached during runtime.

chelmuth, 2022-04-06:
I merged 0b5ea2c60351c44baf60ac640a6f799993d18042 and 756e2e74c6fc7379e1ddd1332bd1db23085c0f58 but [this change](https://github.com/alex-ab/genode/commit/35ccdc5a9192ed27aed913752e6ac31855defcb3#diff-c41e6110bd3651ccb9db9098ec4a78c7a376789bf160e1ddfd9f7eb14f9336a6L60-R60) puzzled me and I did not find any declaration of this "interface" in the further changes below. I'd appreciate if this feature of lx_emul would grow into a real API function with official declaration. I did not investigate all the details, but may it be possible to implement `lx_emul_force_jiffies_update()` as regular function in lx_emul? @skalk, what's your opinion?

skalk, 2022-04-07:
@chelmuth: sorry, I've missed your question. Well, I understand that turning the compilation unit's private `dde_clock_event_device` into a global variable is not welcome.
On the other hand, I cannot overview the effects of this new, rather complex `lx_emul_force_jiffies_update` routine. Actually, I thought that using the same simple procedure that we already used in the `schedule` function would be enough, which looks like the following:
```
tick_nohz_idle_enter();
lx_emul_time_handle();
tick_nohz_idle_exit();
```
I'm certain that @alex-ab had his reasons for not using this approach, but instead implemented the new function. Probably some unwanted side-effects of `lx_emul_time_handle` I'm not aware of?!
To sum it up, I do not really like to have a more complex, additional update jiffie routine within the lx_emul library, but if it is necessary, and as it will be an interim solution anyway (until we replace the whole kernel's timing functions by our own implementation) I can live with. And of course it is possible to move the whole code into `dde_linux/src/lib/lx_emul/clocksource.c`.

chelmuth, 2022-04-08:
If this is feasible, I'd feel much more comfortable with the additional function moved into the official clocksource file. Even if this is an interim solution, we will definitely stumble over it when changing the timing implementation and can consider its requirements.

alex-ab, 2022-04-08:
> tick_nohz_idle_enter();
> lx_emul_time_handle();
> tick_nohz_idle_exit();

I implemented it with 041e58e7143cdd54c9d48b6854995c9c21f9db91. The disadvantage of this version is that on lx_emul_time_handle the jiffies is increased ever, even if no time passed. The handler in kernel/time/timer.c expects to be invoked in a periodic way (called tick_handle_periodic() in kernel/time/tick-common.c). It is more inaccurate. Additional this version reprograms the timeout in clockevents_program_event() in kernel/time/clockevents.c, which is not intended, we need actually just to update the jiffies to current.

skalk, 2022-04-08:
As already stated I do not balk to move it into clockevents.c, and I was sure that @alex-ab had good reasons for his changes. @alex-ab do you proceed with moving the code?

alex-ab, 2022-04-08:
I tested 041e58e7143cdd54c9d48b6854995c9c21f9db91 now on the x250 again, and the wrong timeouts fire again.

alex-ab, 2022-04-08:
> As already stated I do not balk to move it into clockevents.c, and I was sure that @alex-ab had good reasons for his changes. @alex-ab do you proceed with moving the code?

Ok, I will do.

alex-ab, 2022-04-08:
a636e78749f61e75b6999876697a5f4a4c04a9c3 contains the move into the clocksource.c

alex-ab, 2022-04-12:
@chelmuth, @nfeske: please add bc0300d961305a12a0d62a3dbaa000ca42eb908f to staging, it fixes the driver for t430s

commit d0f65ee7 (2022-03-21): pc: update intel display driver

Fixes #4450
commit 1ecfa09c (2022-03-21): fixup "pc: update intel display driver" (wc lx kit change)

Fixes #4450
commit 27aae4c2 (2022-03-25): fixup "pc: update intel display driver" (fix distortion, avoid contradicting framebuffer_register calls)

Fixes #4450
commit bf5fdd48 (2022-03-25): fixup "pc: update intel display driver" (rcu)

Fixes #4450
closed 2022-04-13 by chelmuth