genodelabs/genode #4451: gpu/intel: Vastly improve resource management

ssumpf, 2022-03-22:
Idea: Make worst case assumptions about CAPS and RAM within the Intel GPU multiplexer. Currently this would amount to 4 caps and 2MB of RAM because the Heap (might require an update). Enforce that these quotas are available before any GPU multiplexer operation. Check used quotas before and after allocations and account the actually used resources to a session through  `Quota_guard::withdraw/replenish` for RAM/CAPS respectively. CAPS and RAM will only be given back on `free_buffer` (`replenish`). All other meta data allocation, like page table memory and heap allocations, will remain in the GPU multiplexer until the session is closed. This implies that the GPU multiplexer needs to have enough CAPS/RAM to satisfy a session close request. This is a temporary solution only because the GPU multiplexer will leak resources to the platform driver.

nfeske, 2022-03-22:
The saving sounds exciting. All the more pleasantly surprised I was when looking at the commits, which appear as a quite substantial simplification as well. I merged them to staging and will give them a try as soon as possible.

ssumpf, 2022-03-22:
@nfeske: Interesting. I will look into the mesa-gears case. This should not require more RAM after being started.

ssumpf, 2022-03-23:
@nfeske: Unfortunately I cannot reproduce the `gears` resource request on fuji4 (run for 8 hours over night). This might be a Broadwell problem then, which I will have to test separately. Until then, I will look into the dangling allocations when closing `glmark2`, test vbox6 3D support as well as Doom3 (which is the real world work load that requires thousands of buffers).

ssumpf, 2022-04-05:
@chelmuth, @nfeske: I made a terrible mistake by using `used_caps` and `used_ram` as a basis for the available resources of the GPU multiplexer. These functions do not reflect the actual resources given to the platform driver, which in turn falsifies the whole calculation. This error is fixed by 1cf3bea. Additionally, I observed an endless quota upgrade loop by the legacy platform driver, caused by `Platform::Device_pd::attach_dma_mem` that did not reflect the updated quotas to core. This is fixed by 926904b.

chelmuth, 2022-04-06:
Thanks, I merged both commits to staging.

nfeske, 2022-04-13:
Fixed in master.