genodelabs/genode #4456: memcpy and memset improvements

jschlatow, 2022-03-25:
When comparing memory bandwidth and hunting the culprit behind #4454, I learned about a few angled on how to tweak memcpy and memset performance.

First of all, I started with ~300MiB/s for `Genode::memcpy` and `Genode::memset` on 'a Zynq-7000 SoC @666MHz. The most effective parameters for tuning the throughput were:

* using pld in the `memcpy_cpu()` implemention
* enabling prefetching for L2 and tuning the offset
* enabling double linefills for L2
* changing memory attributes from "write-back write-allocate" to "write-back write no-allocate"

The last point is open for discussion though, as it only holds for the memcpy case in which the written data is not used. This applies to device memory but not in general. Let's have a look at the results with write-back write no-allocate:

```
bytewise:                copied 8 GB in 40142 milliseconds (204 MiB/sec)
memcpy:                  copied 8 GB in 7712 milliseconds (1062 MiB/sec)
memset:                  copied 8 GB in 4000 milliseconds (2048 MiB/sec)
libc memcpy:             copied 8 GB in 20831 milliseconds (393 MiB/sec)
libc memset:             copied 8 GB in 6449 milliseconds (1270 MiB/sec)
memcpy (page offset):    copied 8 GB in 6928 milliseconds (1182 MiB/sec)
memcpy (uncached write): copied 8 GB in 7200 milliseconds (1137 MiB/sec)
memcpy (uncached read):  copied 8 GB in 38959 milliseconds (210 MiB/sec)
```

In contrast, this is what I got without changing the policy:
```
bytewise:                copied 8 GB in 40158 milliseconds (203 MiB/sec)
memcpy:                  copied 8 GB in 12256 milliseconds (668 MiB/sec)
memset:                  copied 8 GB in 4032 milliseconds (2031 MiB/sec)
libc memcpy:             copied 8 GB in 20926 milliseconds (391 MiB/sec)
libc memset:             copied 8 GB in 6448 milliseconds (1270 MiB/sec)
memcpy (page offset):    copied 8 GB in 10159 milliseconds (806 MiB/sec)
memcpy (uncached write): copied 8 GB in 7215 milliseconds (1135 MiB/sec)
memcpy (uncached read):  copied 8 GB in 41567 milliseconds (197 MiB/sec)
```

So, by changing the cache policy, we could get 400MiB/s in this isolated benchmark. Yet, I also find it interesting that offsetting the src and dst buffers by one page gains another 100MiB/s. Placing the dst buffer in uncached memory has a similar effect. For me, this looks a lot like cache pollution.

According to these measurements, a general suggestion (on Cortex-A9) could be to use uncached memory whenever the data is write-only and keep the write-allocate policy since the written data is likely accessed in the near future. Interestingly, Linux uses write-back write no-allocate probably because this reduced cache pollution in any case.

jschlatow, 2022-03-25:
When working on this, I also stumbled upon an apparently obsolete vfp/neon implementation for `memcpy_cpu`. As I expected, it does not improve the memcpy results though since the prefetching seems to be the major bottleneck and ARMs load-store multiple already loads/stores a cache line in 4 cycles if I read the docs correctly. I therefore removed the implementation in commit cac3c33.

chelmuth, 2022-03-29:
Kudos, these are impressing results! I merged all commits in #4454 and this issue to let the CI do the heavy testing on our target platforms.

nfeske, 2022-04-13:
Arrived in the master branch.

closed 2022-04-13 by nfeske