genodelabs/genode #4459: vfs/rump: access to (large?) files on 22.02 leads to unresponsive entrypoint

trimpim, 2022-03-28:
We encountered it first when we were running a virtual machine on our scenario. Init reported skipped heartbeats from the vfs that hosted the rump plugin.                                                                                   
                                                                                                                              
The problem can also be reproduced by copying large (~15GB) files on sculpt (inspect shell).
                                                             
I executed the following tests on my thinkpad. I placed a 15GB image                                                        
on the third partition of AHCI drive and booted Sculpt from USB.                                         
                                                                                                                              
I tested the following commits:                                                                                             
                                                                                                                              
| commit/tag | test result | comment |                                                                                      
| --- | --- | --- |                                                                                                         
| 22.02 | fails | |                                              
| 21.11 | success | |                                                          
| 59ca8f2489 | success| last commit before removing blocking semantics of VFS |
| 2b0e64e061 | fails | os: remove blocking semantics from packet stream |          
| 911ff31709 <br/> 997a24e91a <br/> 88dec4cc94 | all fail | commits changing rump |
                            
The copy operation takes about 4 minutes on my particular disk device. Success means at least 4 file copies did complete.  
                                                                                                                              
In rare cases one single copy was successful on commits that fail most of the time.

nfeske, 2022-03-28:
Thank you very much for the bisecting and reporting. The symptom reminds me of #4433, which was addressed by commit 88dec4cc94b6e86026bb0f7fe0d00afbb8dd64ac.

Your bisecting in between the commits 2b0e64e06160f554e3ff18dba80f015a76d1721f and 88dec4cc94b6e86026bb0f7fe0d00afbb8dd64ac may probably show issue #4433, which may be different from the one you currently observe. To rule this out, could you retry the test with reordering commits 911ff3170927525bad921720945f3a361661c2bd and 88dec4cc94b6e86026bb0f7fe0d00afbb8dd64ac directly on top of 2b0e64e06160f554e3ff18dba80f015a76d1721f (only if quickly possible without too many conflicts of course)?

However, since this commit is part of 22.02, it might be a different problem.

@cproc may you give @trimpim's way of reproducing the issue with Sculpt (inspect) a try and investigate?

trimpim, 2022-03-28:
@nfeske I will do that tomorrow, and keep you updated on my findings.

cproc, 2022-03-28:
I couldn't reproduce the problem with Sculpt on 22.02 and staging on my test machine so far (x260, TFTP boot, whole AHCI SSD formatted as ext2, copying a 15GB file).

trimpim, 2022-03-29:
@nfeske moving the two commits to directly after 2b0e64e did not produce any merge conflicts.

When I copied multiple files with the command `cp -fv test*.img /backup` the following error occurred:

```
[runtime -> ahci-1.3.fs] rump: start = 0, len = 66, fs = /
runtime -> ahci-1.3.fs] rump: panic: ext2fs_alloccg: map corrupted
runtime -> ahci-1.3.fs] rump: rump kernel halting ...
runtime -> ahci-1.3.fs] rump: halted
runtime -> ahci-1.3.fs] Warning: rumpuser_sp_fini called (from 0x109ac43)
runtime -> ahci-1.3.fs] Error: rump panic
```
After this I booted Ubuntu to check the integrity of the file system, it was reported to be clean.

This happened only once in 3 tries to copy multiple files.

There is one difference I noticed, that makes no sense to me. Yesterday nearly always the first file copy failed, Today after reordering the commits the first copy always succeeded, but the second was stuck (or crashed rump).

cproc, 2022-03-29:
When I did the copy tests on 22.02 and staging I noticed that copying the 15GB file took about 2m20s on 22.02 and about 4m5s on staging. I'm currently bisecting this symptom and just got an error message that might be related to @trimpim's issue:

```
[init -> drivers -> dynamic -> ahci_drv] Warning: overlap: pending 2 + 2 (read), request: 3 + 1 (read)                                             
[init -> runtime -> ahci-1.fs] Warning: spurious block-operation acknowledgement                                                                   
[init -> runtime -> ahci-1.fs] Error: Mounting 'ext2fs' file system failed (22)                                                                    
[init -> runtime -> ahci-1.fs] Error: failed to create <rump> VFS node                                                                             
[init -> runtime -> ahci-1.fs] Error:         fs="ext2fs"                                                                                          
[init -> runtime -> ahci-1.fs] Error:         ram="48M"                                                                                            
[init -> runtime -> ahci-1.fs] Error:         writeable="yes"
```

When I "un-inspected" and "re-inspected", this error did not occur again.

nfeske, 2022-03-29:
Maybe e3706837b9bb0a44a483a95c77d171d59a1edd26 interferes with rump fs in strange ways?

trimpim, 2022-03-29:
I also tested with staging. 
First copy did not not finish after 30 minutes or so. After "un-inspecting", "checking" and "re-inspecting" I could use the filesystem again. After this I could copy two files in succession. The time was about the same as @cproc has reported.

cproc, 2022-03-29:
According to git bisect, the slow-down is related to commit 2b8a94a (nova: support extended addresses in FADT). When I reverted to NOVA revision 5d107ddf9befe2473a80e6c4fa94b1eecae9b268 on staging, copying the 15GB file got faster again (4m4s -> 2m18s).

cproc, 2022-03-29:
(I'm still on staging branch 0bc6443 without the latest changes of today)

chelmuth, 2022-03-29:
> According to git bisect, the slow-down is related to commit [2b8a94a](https://github.com/genodelabs/genode/commit/2b8a94aef978781d2e39f7c8de6f33b483d8e524) (nova: support extended addresses in [...]

What a bummer... Could you please instrument NOVA in *src/acpi_fadt.cpp* with `Console::print` to log Acpi_gas values for the pm1a_sts etc. statics with and without the patch?

cproc, 2022-03-29:
@chelmuth:

- without patch:

```
pm1a_sts: 
init(): asid: 1, bits: 16, addr: 1800
pm1a_ena: 
init(): asid: 1, bits: 16, addr: 1802
pm1a_cnt: 
init(): asid: 1, bits: 16, addr: 1804
pm2_cnt: 
init(): asid: 1, bits: 8, addr: 1850
pm_tmr: 
init(): asid: 1, bits: 32, addr: 1808
gpe0_sts: 
init(): asid: 1, bits: 128, addr: 1880
gpe0_ena: 
init(): asid: 1, bits: 128, addr: 1890
```

- with patch:

```
pm1a_sts: 
init(): asid: 1, bits: 16, addr: 1800
init(): asid: 1, bits: 16, addr: 1802
pm1b_sts: 
init(): asid: 0, bits: 0, addr: 0
init(): asid: 0, bits: 0, addr: 0
pm1a_cnt: 
init(): asid: 1, bits: 16, addr: 1804
pm1b_cnt: 
init(): asid: 0, bits: 0, addr: 0
pm2_cnt: 
init(): asid: 1, bits: 8, addr: 1850
pm_tmr: 
init(): asid: 1, bits: 32, addr: 1808
gpe0_sts: 
init(): asid: 1, bits: 0, addr: 1880
init(): asid: 1, bits: 0, addr: 1880
gpe1_sts: 
init(): asid: 0, bits: 0, addr: 0
init(): asid: 0, bits: 0, addr: 0
```

chelmuth, 2022-03-30:
Thanks for the log that supports my thesis of a storm of unwanted ACPI interrupts due to GPEs, which slows down the x260. Unfortunately, my NOVA patch did not apply the correct register block length for `gpe0_sts/ena` and, thus, prevented the enable-bit clearing on platform setup. Please try https://github.com/chelmuth/NOVA/commits/fadt_extended_addr NOVA branch.

cproc, 2022-03-30:
@chelmuth: yes, that branch fixes the copy speed issue. I see about 2 to 3 'tmr_ovf' log messages in 5 seconds.

chelmuth, 2022-03-30:
That's great. @alex-ab could you please merge the NOVA fixup? I'll update the base-nova commits on staging afterwards.

Any news about the original rump_fs issue? I looked into ` Warning: overlap: pending 2 + 2 (read), request: 3 + 1 (read)` and it seems to be just a diagnostic message, that reveals such behavior of clients. ahci_drv implements a retry pattern with the request-stream API to defer the second (overlapping) request until the first request was finished. As there's no test for this special-case, it's still an open question if the implementation is accurate.

cproc, 2022-03-30:
I had the 15GB file being copied in a loop for over an hour without error on the x260 and am going to try other hardware now.

alex-ab, 2022-03-30:
> That's great. @alex-ab could you please merge the NOVA fixup? I'll update the base-nova commits on staging afterwards.

Done, I moved the old branches to *_old and squashed your fixup, see r10_fadt and r10_iommu.

trimpim, 2022-03-30:
@nfeske I used 4591d501b442a22ba5c114dca3cae159bf3758f4 for my tests today.

cproc, 2022-03-30:
Ok, it could have something to do with the way the ```cp``` command is executed. In my previous tests I ran ```cp``` for a single file in a loop, but with @trimpim's  ```cp -fv test* backup/``` command, something gets stuck on the x260 as well. In this situation Genode ```top``` shows no activity and Ctrl-C does not work. Will investigate further.

cproc, 2022-04-01:
Current findings are that the ep thread of the vfs server is executing a read request through rump and while doing that a different thread does a synchronous write and waits for the ep thread to handle the signal, which does not happen for some reason.

```
[init -> runtime -> ahci-1.fs] rump: 0x401fe858: VOP_READ()
[init -> runtime -> ahci-1.fs] rump: 0x401fe858: VOP_READ() finished: 0
[init -> runtime -> ahci-1.fs] rump: 0x401fe858: VOP_READ()
[init -> runtime -> ahci-1.fs] 0x409fe76c: submit(): op: 2, length: 4096
[init -> runtime -> ahci-1.fs] 0x409fe5dc: block_for_io()
```

cproc, 2022-04-04:
When the rump code executed by the ep thread cannot allocate enough memory, it wakes up the "page daemon" thread and waits for it to free some memory. But if the page daemon thread now does synchronous block io and waits for the ep thread to handle the signal, a deadlock occurs. Not sure yet how to fix this.

```
[init -> runtime -> ahci-1.fs] 0x401: rump_hypermalloc(): howmuch: 69632, free: 52476
[init -> runtime -> ahci-1.fs] 0x401: uvm_wait(): waking up pdaemon
[init -> runtime -> ahci-1.fs] 0x401: uvm_wait(): waiting
[init -> runtime -> ahci-1.fs] 0x409: uvm_pageout(): woke up
[init -> runtime -> ahci-1.fs] 0x409: submit(): op: 2, length: 4096
[init -> runtime -> ahci-1.fs] 0x409: block_for_io()
```

chelmuth, 2022-04-05:
That's bad because uvm_wait uses condvar a.k.a the ancient Timed_semaphore, which is not aware of the required blocking scheme like `Io_signal_blockade` does. Do you see any chance to apply the `Io_signal_blockade` scheme to the timed semaphore as well?

cproc, 2022-04-07:
I'm currently testing commits ea9f0aa and a3017a5. The copy test still gets stuck sometimes, but it might be a different reason related to child destruction.

cproc, 2022-04-07:
There appears to be a problem with nested calls into rump code via ```wait_and_dispatch_one_io_signal()``` now. In the following log output rump ran out of memory during a read operation. The ep thread woke up the pdaemon thread to free some memory and then blocked on a condition variable, now with `wait_and_dispatch_one_io_signal()`, which executed the `Rump_factory::_sync()` signal handler of the rump vfs plugin, which called into rump again, did a synchronous write and then blocked on a different condition variable (```Ep_blockade``` pointer 0x401fd6a8), apparently waiting for another thread to unblock it, which never happened for some reason. After a while the `Rump_factory::_sync()` signal handler got executed again (with the previous execution not finished yet) and so on until a stack overflow occurred.

```
[init -> runtime -> ahci-1.fs] 0x401: ext2fs_read()
[init -> runtime -> ahci-1.fs] 0x401: rump_hypermalloc(): howmuch: 69632, free: 11316
[init -> runtime -> ahci-1.fs] 0x401: uvm_wait(): waking up pdaemon
[init -> runtime -> ahci-1.fs] 0x401: uvm_wait(): waiting
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fe148: Ep_blockade::block()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fe148: Ep_blockade::block(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: Rump_factory::_sync() signal handler
[init -> runtime -> ahci-1.fs] 0x401: submit(): op: 2, length: 4096
[init -> runtime -> ahci-1.fs] 0x401: _synchronous_io(): calling block_for_io()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x11848d0: Io_signal_blockade::block_for_io()
[init -> runtime -> ahci-1.fs] 0x401: _synchronous_io(): completed: 0
[init -> runtime -> ahci-1.fs] 0x401: _synchronous_io(): completed: 0
[init -> runtime -> ahci-1.fs] 0x401: this: 0x11848d0: Io_signal_blockade::block_for_io(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x407: uvm_pageout(): woke up
[init -> runtime -> ahci-1.fs] 0x407: uvm_pageout(): for
[init -> runtime -> ahci-1.fs] 0x407: uvm_pageout(): waking up ep
[init -> runtime -> ahci-1.fs] 0x407: this: 0x401fe148: Ep_blockade::wakeup()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fe148: Ep_blockade::_handle_wakeup()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x11848d0: Io_signal_blockade::block_for_io(): wait_and_dispatch_one_io_signal() returned
[init -> runtime -> ahci-1.fs] 0x401: this: 0x11848d0: Io_signal_blockade::block_for_io(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x11848d0: Io_signal_blockade::_handle_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x11848d0: Io_signal_blockade::block_for_io(): wait_and_dispatch_one_io_signal() returned
[init -> runtime -> ahci-1.fs] 0x401: _synchronous_io(): completed: 1
[init -> runtime -> ahci-1.fs] 0x401: this: 0x11848d0: Io_signal_blockade::block_for_io() finished
[init -> runtime -> ahci-1.fs] 0x401: _synchronous_io(): block_for_io() returned
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd6a8: Ep_blockade::block()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd6a8: Ep_blockade::block(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd6a8: Ep_blockade::block(): wait_and_dispatch_one_io_signal() returned
<... repeating ...>
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd6a8: Ep_blockade::block(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: Rump_factory::_sync() signal handler
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd098: Ep_blockade::block()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd098: Ep_blockade::block(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd098: Ep_blockade::block(): wait_and_dispatch_one_io_signal() returned
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd098: Ep_blockade::block(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401fd098: Ep_blockade::block(): wait_and_dispatch_one_io_signal() returned
...
[init -> runtime -> ahci-1.fs] 0x401: this: 0x401efc68: Ep_blockade::block(): calling wait_and_dispatch_one_io_signal()
[init -> runtime -> ahci-1.fs] 0x401: Rump_factory::_sync() signal handler
page fault, pd='init -> runtime -> ahci-1.fs' thread='ep' cpu=0 ip=0x67f22 address=0x401eeff8 stack pointer=0x401ef000 qualifiers=0x6 irUWp reason=1
```

nfeske, 2022-04-07:
I sense the root cause of the problem is the fact that `rump_sys_sync` is called (at https://github.com/genodelabs/genode/blob/master/repos/dde_rump/src/lib/vfs/rump/vfs_rump.cc#L51) from the context of an I/O signal handler (at https://github.com/genodelabs/genode/blob/master/repos/dde_rump/src/lib/vfs/rump/vfs_rump.cc#L850). Because of this hack, we end up in the complicated nesting situation.

In my opinion, a clean solution for triggering a periodic sync would be the use of official rump primitives that are legitimate callers of the `rump_sys_` API. Consequently, the periodic sync belongs into a rump kernel thread. This thread would merely execute `rump_sys_sync` and an rump-kernel's sleep function (`rump_sys_nanosleep` or `rump_sys_clock_nanosleep`?) in a loop. This approach would at least respect the layering of the abstractions. @cproc could you try to follow this direction for a bit?

cproc, 2022-04-11:
Moving the periodic sync into a rump kernel thread appears to work. I had the copy test running for several hours over the weekend with the 4 commits on my issue4459 branch [1] without problems so far. @trimpim: feel free to give it a try.

[1] https://github.com/cproc/genode/tree/issue4459

chelmuth, 2022-04-11:
I merged your patches to staging. Unfortunately, I see reproducible errors when just editing a file on ahci.fs in Sculpt.

```
page fault, pd='init -> runtime -> ahci-2.6.fs' thread='rumpclk0' cpu=0 ip=0x424592e0 address=0x424592e0 stack pointer=0x404feb88 qualifiers=0x14 IrUwp reason=1
```

Have you seen this before?

cproc, 2022-04-11:
I haven't seen this error before and could not reprouce it with a short vim session in the Inspect console. Can you give more details what you are doing?

chelmuth, 2022-04-11:
I traced the pagefault until `rumpkern_sched()` in `rumpuser_clock_sleep()`.

@cproc could you please explain how the `Fifo::Element` in `Timed_sempahore` is dequeued on timeout in `_down_internal()`? Is this missing?

cproc, 2022-04-11:
@chelmuth: yes, that seems to be missing. I'll have a look.

chelmuth, 2022-04-11:
With the following patch the issue did not happen since 10 min.

```patch
 repos/dde_rump/src/include/rump/timed_semaphore.h | 6 +++++-
 1 file changed, 5 insertions(+), 1 deletion(-)

diff --git a/repos/dde_rump/src/include/rump/timed_semaphore.h b/repos/dde_rump/src/include/rump/timed_semaphore.h
index f5cd017666..f85133398c 100644
--- a/repos/dde_rump/src/include/rump/timed_semaphore.h
+++ b/repos/dde_rump/src/include/rump/timed_semaphore.h
@@ -165,8 +165,12 @@ class Timed_semaphore
 			 * */
 			queue_element.blockade.block();
 
-			if (queue_element.timed_out)
+			if (queue_element.timed_out) {
+				Genode::Mutex::Guard guard(_meta_lock);
+				_queue.remove(queue_element);
+
 				throw Timeout_exception();
+			}
 		}
 
 	public:
```

chelmuth, 2022-04-11:
BTW, could we replace the exception by an explicit return value for the cases?

alex-ab, 2022-04-11:
I skimmed the implementation, and have some questions (Sorry if I'm wrong). Since we have here several threads involved, shouldn't setting and checking the timed_out state be protected by the _meta_lock ? In case the timeout did not fire, I believe the timeout must be discard(), otherwise it may still trigger (on an object on the stack which is gone already)?

cproc, 2022-04-12:
@alex-ab: thanks for reviewing. I made some changes with regard to your and @chelmuth's comments in commits 6499e95 and 733a75f.

trimpim, 2022-04-12:
@cnuke I tested 733a75f on my Lenovo T460p. After about 12 files there were the two message below in the log, after which the copy seemed to be stuck again:
```
[runtime -> inspect -> bash] Error: no plugin found for (255)
[runtime -> ahci-1.3.fs] warning: _wait_and_dispatch_one_io_signal called from non-entrypoint thread "genode_sync_thr"
```

I'm not sure if the two are related, or even how far apart they were printed in the log.

chelmuth, 2022-04-12:
Commit [6499e95](https://github.com/cproc/genode/commit/6499e951e912bf15654404cb17d5ee2b1ea5a6e7) introduces a new issue: In https://github.com/cproc/genode/blob/6499e951e912bf15654404cb17d5ee2b1ea5a6e7/repos/dde_rump/src/include/rump/timed_semaphore.h#L139-L144 the mutexes are acquired in reverse order (destruct - meta) as in the context of https://github.com/cproc/genode/blob/6499e951e912bf15654404cb17d5ee2b1ea5a6e7/repos/dde_rump/src/include/rump/timed_semaphore.h#L202-L216 (meta - destruct) which is prone to deadlocks.

Also, I'm not sure about the use of the attempt pattern - it looks quite artificial but maybe I'm not yet used to it.

cproc, 2022-04-12:
Commit ac47845 makes the ep thread pointer a constructor argument of the Timed_semaphore and initializes it in the rump env instead.

cproc, 2022-04-12:
Commit 9cba5da fixes the arm_v8a build.

nfeske, 2022-04-12:
> Is it too much for this use case?

I think it is good. But I think you can simplify this code:
```
return queue_element.block().convert<Down_result>(
				[&] (Down_ok) { return Down_ok(); },
				[&] (Down_timed_out) { return Down_timed_out();	}
);
```

by replacing it with a plain return statement because the type is same, isn't it?
```
return queue_element.block();
```

trimpim, 2022-04-12:
@cproc I am currently running ac4784540c. So far 25 file copies were successful, and it is still running.

cproc, 2022-04-12:
> by replacing it with a plain return statement because the type is same, isn't it?

Right. In a previous version one of the lambdas did more than just return, but not anymore. Fixed by d79dae1.

cproc, 2022-04-12:
Now that the destruct mutex only covers `blockade.wakeup()` and since `blockade` is only a reference member in `Element`, the mutex might fit better in the `Timed_semaphore_*_blockade` classes (if it is needed at all).

trimpim, 2022-04-13:
@cproc do you think it is possible to back port this commits to 22.02 without too much trouble?

https://github.com/genodelabs/genode/commit/ac4784540c54ea35fd067d6907ee3bdafe31a24e has now run for about 20 hours without problem.

chelmuth, 2022-04-13:
Commit ac4784540c54ea35fd067d6907ee3bdafe31a24e differs from our current staging like follows, which includes a fix for potential deadlocks. Also, as all dde_rump related fixup commits will be stashed you should consider to update your sources.

```diff
diff --git a/repos/dde_rump/src/include/rump/timed_semaphore.h b/repos/dde_rump/src/include/rump/timed_semaphore.h
index b4740a6f719..ecf27941d5e 100644
--- a/repos/dde_rump/src/include/rump/timed_semaphore.h
+++ b/repos/dde_rump/src/include/rump/timed_semaphore.h
@@ -138,8 +138,6 @@ class Timed_semaphore
 
 			void handle_timeout(Genode::Duration)
 			{
-				Genode::Mutex::Guard guard(destruct_mutex);
-
 				{
 					Genode::Mutex::Guard guard(meta_mutex);
 
@@ -160,6 +158,12 @@ class Timed_semaphore
 					}
 				}
 
+				/*
+				 * Protect the 'blockade' member from destruction
+				 * until 'blockade.wakeup()' has returned.
+				 */
+				Genode::Mutex::Guard guard(destruct_mutex);
+
 				blockade.wakeup();
 			}
 
@@ -202,8 +206,6 @@ class Timed_semaphore
 			/* meta_mutex must be acquired when calling and is released */
 			void wakeup()
 			{
-				Genode::Mutex::Guard guard(destruct_mutex);
-
 				/*
 				 * It is possible that 'handle_timeout()' is already being
 				 * called and waiting for the meta_mutex, so in addition to
@@ -215,8 +217,19 @@ class Timed_semaphore
 
 				meta_mutex.release();
 
+				/*
+				 * 'timeout.discard()' waits until an ongoing signal
+				 * handler execution is finished, so meta_mutex must
+				 * be released at this point.
+				 */
 				timeout.discard();
 
+				/*
+				 * Protect the 'blockade' member from destruction
+				 * until 'blockade.wakeup()' has returned.
+				 */
+				Genode::Mutex::Guard guard(destruct_mutex);
+
 				blockade.wakeup();
 			}
 		};
@@ -242,10 +255,7 @@ class Timed_semaphore
 			 * waiting for getting woken up from another thread
 			 * calling 'up()' or by the timeout handler.
 			 */
-			return queue_element.block().convert<Down_result>(
-				[&] (Down_ok) { return Down_ok(); },
-				[&] (Down_timed_out) { return Down_timed_out();	}
-			);
+			return queue_element.block();
 		}
 
 	public:
diff --git a/repos/dde_rump/src/lib/rump/sync_thread.c b/repos/dde_rump/src/lib/rump/sync_thread.c
index 5edcb4b0491..8a231d58fd2 100644
--- a/repos/dde_rump/src/lib/rump/sync_thread.c
+++ b/repos/dde_rump/src/lib/rump/sync_thread.c
@@ -11,9 +11,10 @@
  * under the terms of the GNU Affero General Public License version 3.
  */
 
+#include <sys/cpu.h>
 #include <sys/param.h>
-#include <rump/rumpuser.h>
 #include <sys/vfs_syscalls.h>
+#include <rump/rumpuser.h>
 
 extern void rump_io_backend_sync();
 
```

trimpim, 2022-04-13:
@chelmuth sorry I was probably not clear enough.

My intend is to use all the commits related to this issue (88c1869 .. 42cfc82)  to our current branch, until we are ready to move to `22.05` after it is released.

nfeske, 2022-04-13:
The fixes entered the master branch.