genodelabs/genode #4461: Sculpt: depot preparation fails

jschlatow, 2022-03-28:
I've built a sculpt image from the current staging branch and noticed that it fails to prepare the depot correctly. I instrumented the code to get more log output. Apparently, the copy operation from /config/depot to /rw/depot fails for some files:

```
[runtime -> prepare -> vfs] bash: warning: command substitution: ignored null byte in input
[runtime -> prepare -> vfs] (null): cannot stat '/rw/config/21.11/*': No such file or directory
[runtime -> prepare -> vfs] '/config/depot/alex-ab' -> '/rw/depot/alex-ab'
[runtime -> prepare -> vfs] '/config/depot/alex-ab/download' -> '/rw/depot/alex-ab/download'
[runtime -> prepare -> vfs] '/config/depot/alex-ab/pubkey' -> '/rw/depot/alex-ab/pubkey'
[runtime -> prepare -> vfs] (null): will not create hard link '/rw/depot/chelmuth' to directory '/rw/depot/alex-ab'
[runtime -> prepare -> vfs] '/config/depot/cnuke' -> '/rw/depot/cnuke'
[runtime -> prepare -> vfs] '/config/depot/cnuke/download' -> '/rw/depot/cnuke/download'
[runtime -> prepare -> vfs] '/config/depot/cnuke/pubkey' -> '/rw/depot/cnuke/pubkey'
[runtime -> prepare -> vfs] '/config/depot/cproc' -> '/rw/depot/cproc'
[runtime -> prepare -> vfs] '/config/depot/cproc/download' -> '/rw/depot/cproc/download'
[runtime -> prepare -> vfs] '/config/depot/cproc/pubkey' -> '/rw/depot/cproc/pubkey'
[runtime -> prepare -> vfs] '/config/depot/genodelabs' -> '/rw/depot/genodelabs'
[runtime -> prepare -> vfs] '/config/depot/genodelabs/download' -> '/rw/depot/genodelabs/download'
[runtime -> prepare -> vfs] '/config/depot/genodelabs/pubkey' -> '/rw/depot/genodelabs/pubkey'
[runtime -> prepare -> vfs] (null): will not create hard link '/rw/depot/jschlatow' to directory '/rw/depot/genodelabs'
[runtime -> prepare -> vfs] '/config/depot/mstein' -> '/rw/depot/mstein'
[runtime -> prepare -> vfs] '/config/depot/mstein/download' -> '/rw/depot/mstein/download'
[runtime -> prepare -> vfs] '/config/depot/mstein/pubkey' -> '/rw/depot/mstein/pubkey'
[runtime -> prepare -> vfs] '/config/depot/nfeske' -> '/rw/depot/nfeske'
[runtime -> prepare -> vfs] '/config/depot/nfeske/download' -> '/rw/depot/nfeske/download'
[runtime -> prepare -> vfs] '/config/depot/nfeske/pubkey' -> '/rw/depot/nfeske/pubkey'
[runtime -> prepare -> vfs] (null): will not create hard link '/rw/depot/skalk' to directory '/rw/depot/cproc'
[runtime -> prepare -> vfs] '/config/depot/ssumpf' -> '/rw/depot/ssumpf'
[runtime -> prepare -> vfs] '/config/depot/ssumpf/download' -> '/rw/depot/ssumpf/download'
[runtime -> prepare -> vfs] '/config/depot/ssumpf/pubkey' -> '/rw/depot/ssumpf/pubkey'
[runtime] child "prepare" exited with exit value 1
```

Any ideas, why some directories are interpreted as hard links? Inspecting _/config/depot_ actually shows different contents and thus does not confirm the hard links.

nfeske, 2022-03-28:
I have no explanation but the "hardlink" messages look strange. Maybe `cp` does something clever about inode numbers (creating hardlinks when the numbers match) whereas various Genode VFS plugins pull those numbers out of thin air. Could the inode number of a tar entry of the depot_users.tar match an inode number of a directory in /rw/depot/ by chance?

If this ambiguity is the problem, it would be nice to know how to prevent `cp` from looking a inode numbers.

> Interestingly though, I just experimented with using the vfs_import plugin to populate the depot directory with the depot.tar content. In this case, the prepare child copies everything successfully.

This sounds like a nice light-weight alternative to the current prepare step. However, let us best solve the mystery above before.

jschlatow, 2022-03-29:
@nfeske, you're right with suspecting the inode numbers. The issue though are not the inode numbers per-se but actually the device ids. Inode numbers must be unique within a single file system but not between different file systems. Only the pair of device id and inode number is unique across the system.

In this particular case, we have two ram file systems: one at /config, into which depot.tar is unpacked, and one at /ram/depot. These file systems originate from separate components and might have the same device id (taken from the address of the Ram_file_system object). This is NOT an issue though because, if we integrate both into a vfs server, the device id is taken from the instance of the Fs_file_system. Yet, the vfs of the bash component again uses an Fs_file_system to access the consolidated file system provided by the vfs server, which unfortunately replaces the device ids again.

nfeske, 2022-03-29:
Thanks for the detailed analysis. That's an annoying problem because we actually don't have any notion of device ids at the file-system session level.

As interim solution, I'd go for tweaking `cp` to omit the device-id check and just always do the dumb copy. I don't know if the `--reflink` argument can be of help? In the worst case, we'd patch coreutils.

Down the road, I sense that we need to have a fresh take on those pesky inode numbers. It is not the first time they bite us. One direction would be to generate those numbers from the full path of the file. But given that hash collisions might occur, this adds new uncertainties. It also increases the cost of stat operations.

Alternatively, we could go frontal, declaring those numbers as being officially useless on Genode. As a mere measure to keep POSIX programs like `cp` from comparing inode numbers, we could change the libc to return a new number for each stat call based on a counter. So those comparisons will always fail. Since inode numbers are 64-bit values, this counter won't possibly wrap. Assuming that a stat call takes 100us, the counter would take 60 million years to wrap (according to a scientific calculation of mine).

@chelmuth What do you think of the transient inode numbers idea based on a libc-local counter? If this works out, we can even remove the notion of an `inode` from the `File_system::Status` because Genode does not use inode numbers anyway.

jschlatow, 2022-03-29:
Looking at the code, there is no command line switch for that, thus I patched coreutils (see 680bc82).

nfeske, 2022-03-30:
Thanks @jschlatow. I merged your commit to staging.

nfeske, 2022-04-13:
Fixed in master.

closed 2022-04-13 by nfeske

m-stein, 2022-06-21:
I just wanted to add that I experienced a similar problem with vim in Sculpt 22.04. When first opening ``reports/drivers/usb_devices`` and then trying to do ``:e config/usb`` in that vim instance, the program doesn't load the new file but stays with the first one. With ``ls -i`` I can see that the inode numbers of the files are identical.