genodelabs/genode #4462: NIC router: report link state on session construct/destruct

m-stein, 2022-03-29:
The main issue here is that Uplink sessions communicate their link state through the session lifetime and the NIC router doesn't report on Uplink session construction/destruction when ``<report link_state_triggers="yes">`` is configured, although it clearly should.

However, this issue raises another question - shouldn't every network session construction/destruction at the NIC router report when ``<report link_state_triggers="yes">`` is configured? - It's easy to think of a session construction as a "down-up" flank if the session immediately is "up" and of a session destruction as an "up-down" flank if the session was "up". But going a bit further, a report client interested in link state changes might want to know about network session construction or destruction in general as the link, regardless of its state actually appears resp. disappears, which is also a change.

Any opinions?

m-stein, 2022-03-29:
Another detail that came to my eye while inspecting the code around this issue is that the reporting in the NIC router is currently done directly by the code that triggers a report. An implication of this is that a single "router pass" (e.g. the handling of a packet or a reconfiguration) might cause several report updates. A second implication is that we must take care manually that the report updates are not done in an intermediate router state that produce a broken report (e.g. reconfiguration, see methods ``stop_reporting`` and ``start_reporting``).

An alternative approach would be to merely signal the need for a report update and update the report later in a dedicated "router pass". The router would then always be in a consistent state when updating the report and many report changes would be reasonably merged without interrupting functional "router passes". I also think that this would make the code easier to understand.

Any opinions on that as well?

chelmuth, 2022-04-01:
As I already voiced in our offline discussion I appreciate the harmonized link-state handling and also the controlled reporting after the NIC router reached a consistent state only.

commit e1f39c5a (2022-09-13): nic_router: intf. con/destruct reports link state

Creating and destructing an interface was not considered a change of its real
link state as defined in the description of the <report link_state_triggers="">
config attribute in the router's README. In case of Uplink sessions this is
obviously a problem as they communicate their real link state through session
lifetime. But also in case of NIC sessions it's a possible to create an
interface that is immediately "up" after creation or destruct an interface
without its link state going "down" beforehand.

Taking into account also the practical ap [... cut]
closed 2022-09-21 by chelmuth