genodelabs/genode #4466: libdrm: Get rid of dangling allocations when destructed

ssumpf, 2022-03-31:
During exit of glmark2 dangling allocations can observed

`Warning: 3 dangling allocations at allocator destruction time`

ssumpf, 2022-03-31:
* c61bfbd makes sure to remove all meta data blocks but the initial one. When `free_empty_blocks` is called for meta data, new unused regions may be produced. Therefore, we call `_revert_unused_ranges` and `free_empty_blocks` unitl no more unused regions are found
* 637baea makes sure the driver name strings can be freed with `free`
* f5ec7d4 removes all allocated objects when `Drm_call` is destructed

With the commits above `glmark2` does warn about any dangling allocations during exit

chelmuth, 2022-04-01:
@nfeske what is your stance on 1440f572d26ae0638cd48eb8a88430049e421d54 + ac455649aaef357ee63343069522225e2c376f62 ?

nfeske, 2022-04-01:
That looks good to me, even better with the added comment. Thanks @ssumpf and @chelmuth.

commit 439b6601 (2022-04-01): libdrm/iris: free objects in Drm_call destructor

Make sure to remove all buffers, sync objects, and contexts during
destruction.

Fixes #4466
closed 2022-04-13 by chelmuth