genodelabs/genode #4467: coreutils may not build because misssing aclocal-1.15

chelmuth, 2022-03-31:
After a fresh prepared coreutils port `make -C build/x86_64 noux-pkg/coreutils-minimal` and `./tool/depot/create genodelabs/bin/x86_64/coreutils-minimal UPDATE_VERSIONS=1 FORCE=1` may fail like follows.

```
/home/krishna/src/genode.git/contrib/coreutils-0c59a98223ebc29b87276ea0bfcfa8892798cf32/src/noux-pkg/coreutils/build-aux/missing: line 81: aclocal-1.15: command not found
WARNING: 'aclocal-1.15' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
```

`make -d` revealed that timestamps in the port force the rebuild of aclocal macros.

```
        Prerequisite '/home/krishna/src/genode.git/contrib/coreutils-0c59a98223ebc29b87276ea0bfcfa8892798cf32/src/noux-pkg/coreutils/m4/cu-progs.m4' is newer than target '/home/krishna/src/genode.git/contrib/coreutils-0c59a98223ebc29b87276ea0bfcfa8892798cf32/src/noux-pkg/coreutils/aclocal.m4'.
```

This issue is related to #4350 and stems from #4262.

chelmuth, 2022-03-31:
`touch /home/krishna/src/genode.git/contrib/coreutils-0c59a98223ebc29b87276ea0bfcfa8892798cf32/src/noux-pkg/coreutils/aclocal.m4` makes the pkg build again.

nfeske, 2022-04-01:
I wonder, might there be a solution in the lines of 4f820cf1e5d46c4fa6d6cde32a4dd4d3c3b6af0c? I'm unfortunately not well-versed in the autoconf magic, but maybe there exists an equally simple way of discharging the automatism behind the invocation of the aclocal tool?

nfeske, 2022-04-01:
The rule in coreutils's Makefile.in at line 6025 looks like a promising candidate:
```
$(ACLOCAL_M4):  $(am__aclocal_m4_deps)
	$(am__cd) $(srcdir) && $(ACLOCAL) $(ACLOCAL_AMFLAGS)
```
Could you check if prefixing this line with `IGNORE-` has the desired effect? If so, we should extend the `_discharge_automake` command in tool/ports/mk/install.mk by covering this case.

chelmuth, 2022-04-01:
Thanks for the exact pointer. Commit 952ba29c8a0fed0bf2e2888b6a694efefed07f2a fixes the issue (tested with coreutils and tar).

commit 952ba29c (2022-04-01): prepare_ports: prevent regeneration of aclocal.m4

Related to #4350
Fixes #4467
closed 2022-04-13 by chelmuth