genodelabs/genode #4477: pc_usb_host_drv does not work on Thinkpad X201

cnuke, 2022-04-19:
Using the new `pc_usb_host_drv` on the Thinkpad X201 renders USB unusable. The machine contains two EHCI host-controllers where the first one is connected to the USB-A ports on the left hand side and the second one to the USB-A port on the right hand side (docking stations where not checked).

Trying to drive both host-controllers leads to the following:

```
[init -> usb_drv] override physical BDF 0:1a.0 -> 0:1.0
[init -> usb_drv] override physical BDF 0:1d.0 -> 0:2.0
[init -> usb_drv] <devices>
[init -> usb_drv]       <device name="pci-0:1.0" type="pci">
[init -> usb_drv]               <irq number="8"/>
[init -> usb_drv]               <physical_io_mem phys_addr="0xf2728000"/>
[init -> usb_drv]               <io_mem phys_addr="0x40000000" size="0x400" bar="0"/>
[init -> usb_drv]       </device>
[init -> usb_drv]       <device name="pci-0:2.0" type="pci">
[init -> usb_drv]               <irq number="9"/>
[init -> usb_drv]               <physical_io_mem phys_addr="0xf2728400"/>
[init -> usb_drv]               <io_mem phys_addr="0x40001000" size="0x400" bar="0"/>
[init -> usb_drv]       </device>
[init -> usb_drv] </devices>
[init -> usb_drv] 6NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 16
[init -> usb_drv] 6clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[init -> usb_drv] get_random_bytes_arch: leaving buffer unmodified!
[init -> usb_drv] 6clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[init -> usb_drv] 6usbcore: registered new interface driver usbfs
[init -> usb_drv] 6usbcore: registered new interface driver hub
[init -> usb_drv] 6usbcore: registered new device driver usb
[init -> usb_drv] 6PCI: Probing PCI hardware
[init -> usb_drv] 6PCI host bridge to bus 0000:00
[init -> usb_drv] pci_bus 0000:00: root bus resource 11152e0R
[init -> usb_drv] pci_bus 0000:00: root bus resource 11152a0R
[init -> usb_drv] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[init -> usb_drv] pci 0000:00:01.0: [8086:3b3c] type 00 class 0x000c0320
[init -> platform_drv] usb_drv -> : assignment of PCI device 00:1a.0 succeeded
[init -> usb_drv] pci 0000:00:01.0: reg 0x10: 18d6f8R
[init -> usb_drv] pci 0000:00:02.0: [8086:3b34] type 00 class 0x000c0320
[init -> platform_drv] usb_drv -> : assignment of PCI device 00:1d.0 succeeded
[init -> usb_drv] pci 0000:00:02.0: reg 0x10: 18de10R
[init -> usb_drv] pci_bus 0000:00: busn_res: 405fed70R end is updated to 00
[init -> usb_drv] 6clocksource: Switched to clocksource dde_counter
[init -> usb_drv] 6ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[init -> usb_drv] 6ehci-pci: EHCI PCI platform driver
[init -> usb_drv] ehci-pci 0000:00:01.0: EHCI Host Controller
[init -> usb_drv] ehci-pci 0000:00:01.0: new USB bus registered, assigned bus number 1
[init -> usb_drv] ehci-pci 0000:00:01.0: Enabling legacy PCI PM
[init -> platform_drv] 00000000:0000001a.0 adjust IRQ as reported by ACPI: 11 -> 23
[init -> platform_drv] 00000000:0000001a.0 uses IRQ, vector 0x17
[init -> usb_drv] ehci-pci 0000:00:01.0: irq 8, io mem 0x40000000
[init -> usb_drv] ehci-pci 0000:00:01.0: USB 2.0 started, EHCI 1.00
[init -> usb_drv] hub 1-0:1.0: USB hub found
[init -> usb_drv] hub 1-0:1.0: 3 ports detected
[init -> usb_drv] ehci-pci 0000:00:02.0: EHCI Host Controller
[init -> usb_drv] ehci-pci 0000:00:02.0: new USB bus registered, assigned bus number 2
[init -> usb_drv] ehci-pci 0000:00:02.0: Enabling legacy PCI PM
[init -> platform_drv] 00000000:0000001d.0 adjust IRQ as reported by ACPI: 11 -> 19
[init -> platform_drv] 00000000:0000001d.0 uses IRQ, vector 0x13
[init -> usb_drv] ehci-pci 0000:00:02.0: irq 9, io mem 0x40001000
[init -> usb_drv] ehci-pci 0000:00:02.0: USB 2.0 started, EHCI 1.00
[init -> usb_drv] hub 2-0:1.0: USB hub found
[init -> usb_drv] hub 2-0:1.0: 3 ports detected
[init -> usb_drv] 6ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[init -> usb_drv] 6ohci-pci: OHCI PCI platform driver
[init -> usb_drv] 6uhci_hcd: USB Universal Host Controller Interface driver
[init -> usb_drv] 6usbcore: registered new interface driver skel
[init -> usb_drv] get_random_bytes: leaving buffer unmodified!
[init -> usb_drv] 6sched_clock: Marking stable (55365000, 20347000)->(96673000, -20961000)
[init -> usb_drv] hub_port_reset:2939 hub: 1968f8 port1: 1 udev: 19b438
[init -> usb_drv] hub_port_reset:3032 status: 0
[init -> usb_drv] hub_port_reset:2939 hub: 192580 port1: 1 udev: 1a22f0
[init -> usb_drv] usb 2-1: new high-speed USB device number 2 using ehci-pci
[init -> usb_drv] hub_port_reset:3032 status: 0
[... 154 lines cut ...]
[init -> usb_drv] hub_port_reset:3059 status: -16
[init -> usb_drv] hub_port_init:5025 retval: -16
[init -> usb_drv] usb usb1-port1: attempt power cycle
[init -> usb_drv] hub_port_reset:2939 hub: 191e58 port1: 1 udev: 19b880
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[init -> usb_drv] hub_port_reset:3059 status: -16
[init -> usb_drv] hub_port_init:5025 retval: -16
[init -> usb_drv] hub_port_reset:2939 hub: 191e58 port1: 1 udev: 19b880
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] ehci-pci 0000:00:01.0: port 1 reset error -110
[init -> usb_drv] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[init -> usb_drv] hub_port_reset:3059 status: -16
[init -> usb_drv] hub_port_init:5025 retval: -16
[init -> usb_drv] usb usb1-port1: unable to enumerate USB device
```

Through enabling the VGA console and spinner in NOVA I noticed that only interrupts for the first host-controller are received.

cnuke, 2022-04-20:
Commit [75840a1](https://github.com/cnuke/genode/commit/75840a1546fa46814767956928cbfd3b526e3372) addresses the problem where both EHCI host-controller use the same I/O memory page but the registers of the second one are at offsett 0x400.

I opted for adding the offset in `Device::io_mem_local_addr` rather than adapting all `lx_emul_io_mem_map` callees.

m-stein, 2022-04-20:
I added [75840a1](https://github.com/cnuke/genode/commit/75840a1546fa46814767956928cbfd3b526e3372) to my Sculpt 22.04 RC but then USB mouse/keyboard doesn't work anymore (tested with my wired logitech mouse and my wireless cherry mouse/keyboard combo; also dynamically changed slots). Here is the log with the wireless combo (USB dongle is a hub):
```

[drivers -> usb_hid_drv] --- USB HID input driver ---[0m
[drivers -> usb_hid_drv] [34mWarning: use compatibility mode: will claim all HID devices from USB report[0m
[drivers -> usb_hid_drv] Configured HID screen with 0x0 (multitouch=false)[0m
...

[drivers -> usb_drv] override physical BDF 0:14.0 -> 0:1.0[0m

[drivers -> usb_drv] override physical BDF 3a:0.0 -> 0:2.0[0m

[drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)[0m

[drivers -> usb_drv] 6NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 16[0m

[drivers -> usb_drv] 6clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m

[drivers -> usb_drv] get_random_bytes_arch: leaving buffer unmodified![0m

[drivers -> usb_drv] 6clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m

[drivers -> usb_drv] 6usbcore: registered new interface driver usbfs[0m

[drivers -> usb_drv] 6usbcore: registered new interface driver hub[0m

[drivers -> usb_drv] 6usbcore: registered new device driver usb[0m

[drivers -> usb_drv] 6PCI: Probing PCI hardware[0m

[drivers -> usb_drv] 6PCI host bridge to bus 0000:00[0m

[drivers -> usb_drv] pci_bus 0000:00: root bus resource 1117220R[0m

[drivers -> usb_drv] pci_bus 0000:00: root bus resource 11171e0R[0m

[drivers -> usb_drv] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff][0m

[drivers -> usb_drv] pci 0000:00:01.0: [8086:9ded] type 00 class 0x000c0330[0m

[drivers -> platform_drv] usb_drv -> : assignment of PCI device 00:14.0 succeeded[0m

[drivers -> usb_drv] pci 0000:00:01.0: reg 0x10: 18d6f8R[0m

[drivers -> usb_drv] pci 0000:00:02.0: [8086:15c1] type 00 class 0x000c0330[0m

[drivers -> platform_drv] usb_drv -> : assignment of PCI device 3a:00.0 succeeded[0m

[drivers -> usb_drv] pci 0000:00:02.0: reg 0x10: 18de10R[0m

[drivers -> usb_drv] pci_bus 0000:00: busn_res: 405fedb0R end is updated to 00[0m

[drivers -> usb_drv] [31mError: memory-mapped I/O resource 0x40000000 (size=0x10000) unavailable[0m

[drivers -> usb_drv] 6clocksource: Switched to clocksource dde_counter[0m

[drivers -> usb_drv] 6ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver[0m

[drivers -> usb_drv] 6ehci-pci: EHCI PCI platform driver[0m

[drivers -> usb_drv] 6ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver[0m

[drivers -> usb_drv] 6ohci-pci: OHCI PCI platform driver[0m

[drivers -> usb_drv] 6uhci_hcd: USB Universal Host Controller Interface driver[0m

[drivers -> usb_drv] [31mError: memory-mapped I/O resource 0x40000000 (size=0x10000) unavailable[0m

[drivers -> usb_drv] xhci_hcd 0000:00:01.0: init 0000:00:01.0 fail, -14[0m

[drivers -> usb_drv] 4xhci_hcd: probe of 0000:00:01.0 failed with error -14[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: xHCI Host Controller[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: new USB bus registered, assigned bus number 1[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: hcc params 0x200077c1 hci version 0x110 quirks 0x0000000200009810[0m

[drivers -> platform_drv] 0000003a:00000000.0 uses MSI vector 0x7f, address 0xfee00018[0m

[drivers -> usb_drv] hub 1-0:1.0: USB hub found[0m

[drivers -> usb_drv] hub 1-0:1.0: 2 ports detected[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: xHCI Host Controller[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: new USB bus registered, assigned bus number 2[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: Host supports USB 3.1 Enhanced SuperSpeed[0m

[drivers -> usb_drv] hub 2-0:1.0: USB hub found[0m

[drivers -> usb_drv] hub 2-0:1.0: 2 ports detected[0m

[drivers -> usb_drv] get_random_bytes: leaving buffer unmodified![0m

[drivers -> usb_drv] 6sched_clock: Marking stable (25615000, 17669000)->(39303000, 3981000)[0m

[drivers -> platform_drv] dynamic -> intel_gpu_drv -> : assignment of PCI device 00:02.0 succeeded[0m
```

m-stein, 2022-04-20:
Here's a log with my wired mouse:
```
[drivers -> usb_hid_drv] --- USB HID input driver ---[0m

[drivers -> usb_hid_drv] [34mWarning: use compatibility mode: will claim all HID devices from USB report[0m

[drivers -> usb_hid_drv] Configured HID screen with 0x0 (multitouch=false)[0m

...

[drivers -> usb_drv] override physical BDF 0:14.0 -> 0:1.0[0m

[drivers -> usb_drv] override physical BDF 3a:0.0 -> 0:2.0[0m

[drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)[0m

[drivers -> usb_drv] 6NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 16[0m

[drivers -> usb_drv] 6clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m

[drivers -> usb_drv] get_random_bytes_arch: leaving buffer unmodified![0m

[drivers -> usb_drv] 6clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m

[drivers -> usb_drv] 6usbcore: registered new interface driver usbfs[0m

[drivers -> usb_drv] 6usbcore: registered new interface driver hub[0m

[drivers -> usb_drv] 6usbcore: registered new device driver usb[0m

[drivers -> usb_drv] 6PCI: Probing PCI hardware[0m

[drivers -> usb_drv] 6PCI host bridge to bus 0000:00[0m

[drivers -> usb_drv] pci_bus 0000:00: root bus resource 1117220R[0m

[drivers -> usb_drv] pci_bus 0000:00: root bus resource 11171e0R[0m

[drivers -> usb_drv] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff][0m

[drivers -> usb_drv] pci 0000:00:01.0: [8086:9ded] type 00 class 0x000c0330[0m

[drivers -> platform_drv] usb_drv -> : assignment of PCI device 00:14.0 succeeded[0m

[drivers -> usb_drv] pci 0000:00:01.0: reg 0x10: 18d6f8R[0m

[drivers -> usb_drv] pci 0000:00:02.0: [8086:15c1] type 00 class 0x000c0330[0m

[drivers -> platform_drv] usb_drv -> : assignment of PCI device 3a:00.0 succeeded[0m

[drivers -> usb_drv] pci 0000:00:02.0: reg 0x10: 18de10R[0m

[drivers -> usb_drv] pci_bus 0000:00: busn_res: 405fedb0R end is updated to 00[0m

[drivers -> usb_drv] [31mError: memory-mapped I/O resource 0x40000000 (size=0x10000) unavailable[0m

[drivers -> usb_drv] 6clocksource: Switched to clocksource dde_counter[0m

[drivers -> usb_drv] 6ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver[0m

[drivers -> usb_drv] 6ehci-pci: EHCI PCI platform driver[0m

[drivers -> usb_drv] 6ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver[0m

[drivers -> usb_drv] 6ohci-pci: OHCI PCI platform driver[0m

[drivers -> usb_drv] 6uhci_hcd: USB Universal Host Controller Interface driver[0m

[drivers -> usb_drv] [31mError: memory-mapped I/O resource 0x40000000 (size=0x10000) unavailable[0m

[drivers -> usb_drv] xhci_hcd 0000:00:01.0: init 0000:00:01.0 fail, -14[0m

[drivers -> usb_drv] 4xhci_hcd: probe of 0000:00:01.0 failed with error -14[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: xHCI Host Controller[0m

[drivers -> usb_drv] xhci_hcd 0000:00:02.0: new USB bus registered, assigned bus number 1[0m
```

cnuke, 2022-04-21:
You right, the second time I mismatched `bar()` vs. `base()` - apparently it is not an issue on the X250 that I tested besides the X201. 
Sorry for the inconvience @m-stein, commit [650475f](https://github.com/cnuke/genode/commit/650475fac59572bcad2aae44aad0ae3fa6dd93ad) contains the fix up.

cnuke, 2022-04-25:
@chelmuth please merge commit [dcbfc21](https://github.com/cnuke/genode/commit/dcbfc21e39a10c6210fe6913407ed2c2e117c0f6) and thanks to @skalk for the explanation.

nfeske, 2022-04-26:
Commit https://github.com/genodelabs/genode/commit/12e717da2d6bb14e1598501b09e56bd46515796c restores the support for requesting I/O at a window beyond the start of the MMIO ranges. This fixes the boot of Sculpt on the x260.

nfeske, 2022-04-26:
> This fixes the boot of Sculpt on the x260.

Works now on the x201, too.

commit 8f1e8be9 (2022-04-25): lx_kit: map I/O mem with proper offset

In case multiple devices shared the same memory-mapped I/O page
we have to take the offset into account. This commit fixes the
'pc_usb_host_drv' on the Thinkpad X201 where the two EHCI host
controller share the some I/O memory page.

Fixes #4477.
commit 2fd2b7d6 (2022-04-28): lx_kit: map I/O mem with proper offset

In case multiple devices shared the same memory-mapped I/O page
we have to take the offset into account. This commit fixes the
'pc_usb_host_drv' on the Thinkpad X201 where the two EHCI host
controller share the some I/O memory page.

Fixes #4477.
closed 2022-04-28 by chelmuth