genodelabs/genode #4480: Using Genode + seL4 as a secure bootloader

arturkow2000, 2022-04-22:
Hello

We are interested in using seL4 + Genode as a secure OS bootloader. We have
already asked seL4 community [here](https://github.com/seL4/seL4/issues/832),
but we are interested in Genode due to its drivers. From what we have seen
Genode already provides USB host and EEM driver. How about TPM driver?

I remember that seL4 support used to be incomplete and Genode had problems with 
running many components. What is the state os seL4 support currently? Could it
serve as a reliable OS for seL4?

nfeske, 2022-04-25:
Thank you for your interest in Genode!

For inquiries like this, please consider consulting the mailing list (https://genode.org/community/mailing-lists) in the future. The issue tracker primarily focused on technical work topics.

Even though we continue maintaining Genode's existing seL4 support for x86_32, x86_64, and arm_v7a, we have not scheduled any further sel4-related topics on our road map (https://genode.org/about/road-map) right now.

> How about TPM driver?

Genode does not offer a TPM driver.

> What is the state of seL4 support currently?

Genode/seL4 is part of our nightly testing routine, including the comprehensive depot_autopilot test suite. So it is generally in a healthy state. However, in contrast to NOVA or the base-hw kernel, we have not used seL4 in real-world scenarios outside the lab.

I hope that I could answer your questions. If so, I'd appreciate you closing this issue.

arturkow2000, 2022-04-27:
Thank you for quick response. So far it does answer my questions. If I have any more questions I will reach you again.

closed 2022-04-27 by arturkow2000