genodelabs/genode #4487: AR8131 Gigabit Ethernet support

Fhiss, 2022-04-26:
Could you add Qualcomm Atheros AR8131 Gigabit Ethernet support?

nfeske, 2022-04-27:
Genode's network drivers are based on the drivers of the iPXE boot loader (https://ipxe.org/). Even though the iPXE code contains various references to Atheros AR9* devices, your device is apparently not covered. Hence, I see no straight-forward way to support the device with the current version Genode.

If you chose to invest your time into further exploration/experimentation/porting, you may consider trying to port the Linux-based network driver, following the footsteps of the drivers found at repos/pc/. If you don't, I'd appreciate you closing this issue because I see none of the core Genode developers picking it up.

nfeske, 2022-04-29:
> One of the milestones for May will be updated drivers for PC hardware [...]

The NIC driver is not covered by this line of work though. There is currently no development activity in this area.

> An even better option would be to ask you to teach me [...]

That is very cool to read! Thanks for your willingness to invest your own time.

I don't want to raise false expectations regarding the amount of mentoring we are able to provide.

Do you already have a development setup where you can boot Genode on your test hardware separate from your (Linux-based) development machine? That is, booting Genode on your test hardware via TFTP, and getting hold of Genode's log messages over serial or AMT (if supported by your test hardware). If not, setting up such a workflow would be a crucial first step. The following two articles may hopefully serve you as a guide.

* https://genodians.org/valerius/2019-03-12-remote-booting-of-x86-machines-with-pxe-and-intel-amt
* https://genodians.org/chelmuth/2019-01-16-test-machine
* https://genode.org/documentation/genode-foundations/21.05/development/System_integration_and_automated_testing.html

Fhiss, 2022-04-29:
I have a desktop on which I want to install Genode and use it as the only operating system. Now I use Linux on it to perform everyday tasks, but your operating system is very interesting. I did it, but I can't download any packages, and the basic configuration of Sculpt OS doesn't even seem to include a text editor.

By the way, I created the following query: https://github.com/ipxe/ipxe/issues/637

NiKiZe, 2022-04-30:
Please do not tell people to port Linux kernel drivers to iPXE, that is a bad way to get drivers into iPXE (or rather a certain way to not get them into iPXE)

nfeske, 2022-05-02:
> Please do not tell people [...]

I'm afraid this is a misunderstanding. In my initial comment (https://github.com/genodelabs/genode/issues/4487#issuecomment-1111024524) I did not mean to suggest any change to iPXE. Noticing that the device is not supported by iPXE, I contemplated the idea of porting the Linux driver to Genode - as an alternative to our existing iPXE-based driver component. There exists a growing set of Genode device drivers ported from Linux at repos/pc/ (https://github.com/genodelabs/genode/tree/master/repos/pc) such as Intel wireless, Intel graphics, USB. Going forward, it may be sensible to base Genode's NIC drivers on Linux, too.

Seeing how my comment was prone to misinterpretation, I'm sorry for not having been clear enough.

nfeske, 2022-08-17:
I'm closing this issue due to inactivity.

closed 2022-08-17 by nfeske