genodelabs/genode #4498: allocator_avl: fix dangling-allocations warning

m-stein, 2022-05-04:
Since I have worked on #2953, I carry around commit ``allocator_avl: fix dangling-allocations warning`` with my merge_to_staging branch. The original issue description reads:

> ... There is a problem left in the SLAB with falsely reported dangling allocations ... its in the Allocator AVL in _revert_allocations_and_ranges where dangling_allocations is raised also for allocations that contain metadata of the metadata allocator (if the metadata alloc uses the AVL allocator as metadata alloc). These allocations can't be freed up to this point, but its fine anyway. So the warning is misleading. For the Allocator_avl_tpl, the metadata allocator is a Tslab. I added a method virtual bool Allocator_avl::_metdata_of_md_alloc(addr) { return false; } which in Allocator_avl_tpl returns _metdata.metadata(addr) which is implemented in the Slab that walks through all its blocks and compares them with the address. But it didn't work out as expected so far.

I have run some scenarios and the commit doesn't seem to break stuff. However, I recall that it didn't fix all warnings and anyway, there is no test, so far, that proves its necessity / efficiency. I'd love to get rid of the commit either by merging or removing it. Any opinions on whether I should investigate the problem further or dismiss it?

chelmuth, 2022-05-04:
Did you check that da1ef670642376a12441bd8997b34bb028da7cee @ssumpf developed during his work on #4466 does not already solve the original issue?

m-stein, 2022-05-04:
@chelmuth That looks great - so I can finally remove my commit :) Thanks for pointing me to the other commit!

closed 2022-05-04 by m-stein