genodelabs/genode #4500: pc: consolidate LX emul for current drivers

cnuke, 2022-05-06:
The following commits consolidate the currently existing drivers in the *pc* repository by moving the deplicated code from the various dummy implementations and to some degree each *lx_emul.c* into the `pc_lx_emul` library. Changes that would change or rather suplement the *dde_linux* repository are for now also located in this library (the changes should be merged in a following step). While there the various *dep.list* files were decluttered.

So far the commits were tested successfully on a `X250` ( *x86_64* and *x86_32*) including bulding and running Sculpt (to check for working depot archives).

nfeske, 2022-05-09:
From my perspective, the change looks excellent. The line statistics of commit https://github.com/cnuke/genode/commit/c3968a24f4ec004e808156508351d997ecb31fac speaks for itself.

I especially appreciate your choice of a very fine granularity of the building blocks in lx_emul/shadow/, and the attention to detail wrt. file headers and coding style. Thank you!

The only question I see to discuss is whether we should eventually promote some of the lx_emul/shadow/ files to the dde_linux repository. But the selection and relocation of files could very well be a separate step, should be decide so.

I think the fixed label well deserved. Now I'd like to pass the microphone to @skalk. ;-)

skalk, 2022-05-11:
@cnuke thank you for this complex reworking! I like the changes very much. Apart of the 3 comments I did inline for the last commit I do not have any objections against merging it to staging.

skalk, 2022-05-11:
@cnuke ok, I understand the reasoning behind not duplicating the pathes in between repos/pc and repos/dde_linux, and I agree it is better to join the 3 related files into the existing ones in repos/dde_linux as soon as the consolidation with the existing ARM drivers takes place. @nfeske @chelmuth: so I'm fine with the commits entering staging.

cnuke, 2022-05-17:
@chelmuth please merge [eb141ba](https://github.com/cnuke/genode/commit/eb141ba5d17ee0ca623021b46d6af9695d1b88d5) to staging as it fixes nightly CI with the 6235 card.