genodelabs/genode #4505: Explorative streamlining of PCI support for ported Linux drivers

cnuke, 2022-05-12:
When PCI support was initially added to new *DDE Linux*, bus and device as well as resource management were performed by the ported 
Linux kernel. Each driver relied on the following contrib sources:
```
arch/x86/pci/legacy.c
drivers/pci/access.c
drivers/pci/bus.c
drivers/pci/host-bridge.c
drivers/pci/msi.c
drivers/pci/pci-driver.c
drivers/pci/pci-label.c
drivers/pci/pci-sysfs.c
drivers/pci/pci.c
drivers/pci/probe.c
drivers/pci/rom.c
drivers/pci/search.c
drivers/pci/setup-res.c
drivers/pci/slot.c
lib/pci_iomap.c
```

 The *lx_emul/spec/x86/pci.c* layer operated on a low level where `pcibios_scan_root()` was the hook to trigger the population of the PCI bus and the kernel enumerated the bus itself. The following commit gets rid of all of the contrib source but `drivers/pci/pci-driver.c` that is used to register the PCI bus at the kernel's device component framework and provides the implementation of all PCI API related functions used by the various drivers in *pc/lx_emul/pci.c*. In addition to the already restrictive access to the PCI config space handled by *lx_kit/spec/x86/platform.cc* it features its own allow-list extended for the Intel framebuffer and wireless driver (see `_pci_read_config()` and `_pci_write_config()`).

With this list the all drivers were successfully tested on a Thinkpad X250, a Fujitsu Lifebook U7411 and the `pc_usb_host_drv` additionally on an older Lenovo workstation. Judging by the various specification documents there are more config registers that could by accessed (r/w) by a driver so it is by no means a complete list but only snapshot of the machines that were tested under working conditions.

The drivers themself probe physical device resources via the usual `pci_resource_*` macros which is why those are set up in the usual way. Since the legacy platform driver glue is also used here these are already virtual but some driver used hard-coded physical BARs which we have to account for one way or the other.

cnuke, 2022-05-12:
Commit [3a49243](https://github.com/cnuke/genode/commit/3a492435433d9aa060fee566bcff29e8b3b15ad5) contains the changes and could serve as an illustration for adapting the new platform driver.

skalk, 2022-08-30:
This issue is meanwhile obsolete due to work on issue #4578, therefore I close it.

closed 2022-08-30 by skalk