genodelabs/genode #4510: lx_emul: wrong include path for arm_64

skalk, 2022-05-18:
During the port of wireguard this issue came up. Unfortunately during the consolidation of different architectures this issue slipped into. In contrast to x86, where x86_64 also includes `arch/arm/include` this is different for arm_64. Therefore, we also do not have to duplicate our shadow pathes. Moreover, the x86-only lx_kit path used for the platform_session wrapper is not needed for ARM in general.

m-stein, 2022-05-19:
@skalk Unfortunately, commit 86b8ab543d breaks the build of wireguard on x86_64.

m-stein, 2022-05-19:
```
$ make app/wireguard
checking library dependencies...
Skip target app/wireguard/spec/x86_64 because it requires x86_64
  Library ldso_so_support
  Library ld
  Library base
  Library net
  Library jitterentropy
  Library lx_emul
  Library virt_linux_generated
  Library wireguard
    COMPILE  drivers/net/wireguard/device.o
    COMPILE  drivers/net/wireguard/netlink.o
    COMPILE  lx_emul/virt_to_page.o
/home/lypo/genodelabs/genode/repos/dde_linux/src/lib/lx_emul/virt_to_page.c: In function ‘lx_emul_virt_to_pages’:
/home/lypo/genodelabs/genode/repos/dde_linux/src/lib/lx_emul/virt_to_page.c:32:5: error: ‘struct page’ has no member named ‘virtual’
   32 |    p->virtual = (void*)((uintptr_t)page_aligned_virt + i*PAGE_SIZE);
      |     ^~
compilation terminated due to -Wfatal-errors.
make[2]: *** [/home/lypo/genodelabs/genode/repos/base/mk/generic.mk:58: lx_emul/virt_to_page.o] Error 1
make[1]: *** [var/libdeps:144: wireguard.lib] Error 2
make: *** [Makefile:326: gen_deps_and_build_targets] Error 2
```

m-stein, 2022-05-19:
@skalk Sorry, I meant on arm_v8a!

skalk, 2022-05-19:
@m-stein well, my commit was build actually before wireguard entered staging, and the former wrong and additional include pathes introduced errors, and of course might have hidden a problem. I think the latter is the case here for wireguard. Anyway, I'll have a look at it.

skalk, 2022-05-19:
@m-stein sorry my judgement was overhasty, I think I found the problem in my commit.

m-stein, 2022-05-19:
Just tell me if I shall look at it instead.

skalk, 2022-05-19:
@chelmuth there is a fixup commit baa03d0f51ee4e49040b170e711e9c2715f3653e available, which simplifies the organization of the Linux kernel headers we shadow in dde_linux 1:1 as we discussed it today offline.

chelmuth, 2022-05-19:
I merged the commit just now. Could we remove the commented code in *spinlock.c*?

skalk, 2022-05-19:
Sorry, that was not intended. I've added a fixup to remove the comments to my merge_to_staging branch.

closed 2022-05-25 by chelmuth