genodelabs/genode #4511: Consolidate USB test scenarios

skalk, 2022-05-18:
For some time the USB driver based run-script tests got frayed more and more. Now, that the implementation and configuration of USB host controller drivers have even been distributed across different repositories, the situation became even worse. Lately the tests for e.g. i.MX8 Quad EVK were not part of the nightly test-routines for this reason.
To re-integrate platforms with their own repository, and to make maintainance of these run-scripts feasible, a board-specific depot package to test USB clients only(!!!) would be helpful. To not mix it up with productive packages it should be named [board]-test_usb_host_drv.

chelmuth, 2022-05-19:
Thanks for the effort, the first take already looks promising. Regarding the *more complex* run scripts I'd like to lower your expectations as IMO this approach addresses the *test-usb-only* cases. I'd expect *netperf_\*_usb* fits this category but not *vbox\**.

Also, I feel the naming of the packages contradicts the current pattern to *append* the board to package names. What do you think of *test_usb_host_drv-[board]*?

skalk, 2022-05-19:
@chelmuth I did not really thought about using the same approach in the vbox5 scenario, but was looking where else the usb_host_drv + platform_drv stack is used directly resp. by platform.inc. However, I think the smartcard and netperf usb test-cases should be transformed too before marking this issue as fixed.
Regarding the naming scheme, I can change it accordingly.

skalk, 2022-05-19:
@chelmuth I've updated my commit to 24e4975 using the naming scheme like you've proposed.

chelmuth, 2022-05-19:
I merged the commit to staging.

nfeske, 2022-05-25:
@skalk is this issue fixed by https://github.com/genodelabs/genode/commit/f652657d9dca0cab60e26435b5619c7dcbd8edc7 or do you plan working on subsequent steps?

skalk, 2022-05-25:
@nfeske as stated before in this issue, I see smartcard.run and netperf_*_usb.run as part of this issue. I do not necessarily have seen me as the only person responsible to do so, but if this is the case then: yes I plan these subsequent steps.

nfeske, 2022-09-23:
Merged to staging.

nfeske, 2022-10-12:
The commits entered the master branch by now.

closed 2022-10-12 by nfeske