genodelabs/genode #4512: usb_host: busy hang on usb storage plug/unplug

alex-ab, 2022-05-19:
On Sculpt 22.04 the usb_host drivers hang up very busy when plug-in/out an USB storage device assigned to a VM. The overall system is afterwards close to unusable.

[image]

chelmuth, 2022-05-19:
I'm quite curious what happens here, so I extracted the story from the picture below.

```
usb 1-2: new high-speed USB device number 11  (connect    usb-1-11)
usb 2-2: USB disconnect, device number 3      (disconnect usb-2-3)
usb 2-2: new SuperSpeed USB device number 4   (connect    usb-2-4)
usb-2-4.drv initializes device
Some string descriptors are not readable
usb 2-2: USB disconnect, device number 4      (disconnect usb-2-4)
(usb-2-4.drv got stuck?)
usb-2-4.part_block got stuck
```

Now I'd like to have a look at a successful plug of the same device.

alex-ab, 2022-05-19:
```
[init -> usb_drv] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[init -> usb_block_drv] Device plugged
[init -> usb_drv] Could not read string descriptor index: 0
[init -> usb_drv] Could not read string descriptor index: 0
[init -> usb_block_drv] Found USB device: Cra (VygrG .) block size: 512 count: 61800448
[init -> report_rom] report 'usb_block_drv -> devices'
[init -> report_rom]   <devices>
[init -> report_rom]    <device vendor="Corsair " product="Voyager GT 3.0  " block_count="61800448" block_size="512" writeable="false"/>
[init -> report_rom]   </devices>
[init] child "usb_block_drv" announces service "Block"
[init -> usb_block_drv] new block session: tx_buf_size=8388608, cap_quota=5, ram_quota=8402125, diag=0, label="block_tester -> "
[init -> usb_drv] usb 2-1: USB disconnect, device number 2
[init -> usb_drv] xhci_hcd 0000:00:01.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[init -> usb_block_drv] Error: complete error: packet not succeded
[init -> usb_block_drv] Error: request pending: tag: 223 read: 1 buffer: 0x11207200 lba: 454656 size: 1048576
[init -> block_tester] Error: processing READ block=454656 count=2048 failed
[init -> block_tester] finished sequential rx:456704 tx:0 bytes:233832448 size:1048576 bsize:512 duration:1908 mibs:116.876310 iops:116.876310 triggered:224 result:failed
[init -> block_tester] Error: ID space not empty at destruction time
[init -> block_tester] Warning: 4 dangling allocations at allocator destruction time
[init -> block_tester] --- all tests finished ---
[init] child "block_tester" exited with exit value 1
no RM attachment (READ pf_addr=0x4 pf_ip=0x10a6edc from pager_object: pd='init -> usb_drv' thread='ep') 
page fault, pd='init -> usb_drv' thread='ep' cpu=0 ip=0x10a6edc address=0x4 stack pointer=0x40cfe3e8 qualifiers=0x4 irUwp reason=1
```

The usb_block debug commit causes an usb_drv pagefault on unplug of a usb storage device. 
The code line points to https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/lib/lx_emul/usb.c#L459, where the actconfig pointer is nullptr. What is the appropriate way to handle this case ?

chelmuth, 2022-05-19:
The old *raw.cc* skipped operations if `actconfig` was not set.

nfeske, 2022-05-19:
Not sure if related, but with Sculpt 22.04 on the frame.work laptop, the USB host driver triggers a page fault at 0x106bce4 when starting the usb_webcam scenario. The webcam driver finds no match (640x480 at 15 fps is not available) and presumably closes the session. The page fault occurs shortly after.

nfeske, 2022-05-19:
Btw, the good news is that the USB webcam works on the frame.work laptop if I change the usb_webcam driver's config to 30 fps.

nfeske, 2022-05-25:
Is this issue solved by commit https://github.com/genodelabs/genode/commit/115ac58fd0df7a24a9e846987ff4a97349bf2e20?

alex-ab, 2022-05-25:
I could not reproduce it with the mentioned commit, so resolved for now.

closed 2022-05-25 by alex-ab