genodelabs/genode #4519: wireguard: sculpt package

m-stein, 2022-05-26:


m-stein, 2022-05-26:
Fixed:
640d78914c wireguard: sculpt package
ff883184bc wireguard: generic target.mk
6aabc0a1c4 wireguard: rename wireguard lib

chelmuth, 2022-05-27:
While building with the latest commits, the following log messages got my attention.

```
checking library dependencies...
  Library timeout
  Library startup-nova
  Library ldso_so_support
  Library cxx
  Library net
  Library jitterentropy
  Library lx_emul
  Library virt_linux_generated
  Library base-nova-common
  Library base-nova
  Library ld-nova
  Library ld
  Library base
  Library wireguard_lx_inc_dirs
    COMPILE  drivers/net/wireguard/device.o
    COMPILE  drivers/net/wireguard/netlink.o
    MERGE    wireguard_lx_inc_dirs.lib.a
  Library wireguard
  Program app/wireguard/wireguard
    LINK     wireguard
```

The files are always rebuilt because of the following directives in the mk file.

https://github.com/genodelabs/genode/blob/312dde383e5a43d0ca025ca3221fdd4a6780f4d5/repos/dde_linux/lib/mk/wireguard_lx_inc_dirs.inc#L69-L74

May you consider fixing this by applying a correct dependency chain?

chelmuth, 2022-05-27:
Building arm_v8a produces the following warning in the Genode C API for wireguard.

```
checking library dependencies...
  Library lx_emul
  Library virt_linux_generated
  Library wireguard_lx_inc_dirs
    COMPILE  drivers/net/wireguard/device.o
    COMPILE  drivers/net/wireguard/netlink.o
    COMPILE  wireguard.o
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c: In function ‘genode_wg_net_device’:
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c:178:9: warning: taking address of packed member of ‘struct genode_wg_net_device’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  178 |  return &_genode_wg_net_dev.public_data;
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c: In function ‘_genode_wg_config_add_dev’:
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c:232:3: warning: taking address of packed member of ‘struct genode_wg_net_device’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  232 |   &_genode_wg_net_dev.public_data, &_genode_wg_net_dev.pcpu_refcnt);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c:268:3: warning: taking address of packed member of ‘struct genode_wg_net_device’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  268 |   &_genode_wg_net_dev.public_data);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c: In function ‘lx_user_init’:
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c:455:34: warning: taking address of packed member of ‘struct genode_wg_net_device’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  455 |  _genode_wg_rtnl_link_ops->setup(&_genode_wg_net_dev.public_data);
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.../genode/repos/dde_linux/src/app/wireguard/spec/arm_64/../../genode_c_api/wireguard.c:460:3: warning: taking address of packed member of ‘struct genode_wg_net_device’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  460 |   &_genode_wg_net_dev.public_data,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    MERGE    wireguard_lx_inc_dirs.lib.a
```

m-stein, 2022-05-29:
@chelmuth Thanks for the detailed feedback! I will deal with this next.

m-stein, 2022-05-30:
I'm unable to find a satisfying solution for the dependency-chain problem.

CC_OPT_% is set automatically for each contrib file by import-lx_emul_common.inc . However, the resulting modname value - which is the filename without extension (for example "device" for wireguard/device.c) - is not expected by WireGuard code. It rather expects "wireguard" for all WireGuard files. For two WireGuard files, this is critical and for those we want to override CC_OPT_%. 

However, if I try to do without the phony target, I can't get the override to happen after the import file has initialized the variables but before the object files are built. It is normally done too early. I also feel that producing the correct modname right in the import file would be much cleaner but cannot estimate the implications on other ports.

chelmuth, 2022-05-30:
Could [target-specific variables](https://www.gnu.org/software/make/manual/html_node/Target_002dspecific.html) help to solve this?

```
drivers/net/wireguard/device.o: CC_OPT_drivers/net/wireguard/device = -DKBUILD_MODFILE='"wireguard"' -DKBUILD_BASENAME='"wireguard"' -DKBUILD_MODNAME='"wireguard"' 
```

m-stein, 2022-05-30:
@chelmuth I really appreciate your thorough reviews! I've added commits regarding your objections:
a826368107 wireguard: no warning about possible alignm. issue
75f9bfef0c wireguard: don't always rebuild device.c/netlink.c

commit 312dde38 (2022-05-27): wireguard: sculpt package

Introduces the pkg/wireguard archive that depends on the new src/wireguard
archive and deploys one WireGuard component with an empty configuration (can
be configured by the user via a launcher, for instance).

Fixed #4519
commit aea993b9 (2022-05-31): wireguard: sculpt package

Introduces the pkg/wireguard archive that depends on the new src/wireguard
archive and deploys one WireGuard component with an empty configuration (can
be configured by the user via a launcher, for instance).

Fixed #4519
closed 2022-05-31 by chelmuth