genodelabs/genode #4520: wireguard: exit on invalid configurations

m-stein, 2022-05-27:
Currently, the WireGuard component prints an error on invalid configurations but proceeds as usual. Furthermore, the component simply ignores attempts to re-configure attributes that are not re-configurable. Both issues can become a significant security risk in real-world scenarios as the user is misguided to believe that his setup is just fine while in fact it caused WireGuard to operate on an invalid our outdated configuration.

chelmuth, 2022-05-27:
Commit 5ee5845b5de3bf66734f6da175956f29c05000b7 cannot be applied to staging (312dde383e5a43d0ca025ca3221fdd4a6780f4d5). Please rebase for merge.

m-stein, 2022-05-29:
@chelmuth I'm sorry for the inconvenience, I couldn't finish the work on Friday and accidentily pushed a single commit. The issue was actually about several fixes and a test for the wireguard configuration. The correct commit series is this:
3304f444a9 wireguard: enable WARN_STRICT_CONVERSION in lib
8df5fb8f56 wireguard: test multiple peers at a time
71830681a4 wireguard: exit on invalid configurations
59d06b7dd1 wireguard: support removing peers
128a1c12cc wireguard: flags value via enum when adding a peer
60eca7ef57 root/component.h: warnings on Service_denied

nfeske, 2022-08-17:
The commits referencing this issue entered the master branch but the issue lacks the "fixed" label.

@m-stein what is the status of the issue? Should it be closed?

m-stein, 2022-08-17:
It can be closed.

closed 2022-08-17 by m-stein