genodelabs/genode #4522: lx_emul: set task 'stack' member

cnuke, 2022-06-02:


cnuke, 2022-06-02:
Commit [cc8072f](https://github.com/cnuke/genode/commit/cc8072fc392ddd480ec773115289d978a9a94545) sets the 'stack' member to the actual stack of the task rather than allocating member for the thread info.

jschlatow, 2022-07-27:
@cnuke unfortunately, the commit is incompatible with arm (32bit) because it does not select `CONFIG_THREAD_INFO_IN_TASK` but actually expects a `struct thread_info` behind the `stack` member. In this configuration, we must therefore initialise the `stack` member accordingly before calling `task_thread_info()`. I have pushed a fixup with 3d0f9e3. Do you have a more elegant idea?

cnuke, 2022-07-28:
@jschlatow thanks for the fixup. From my point of view your solution is fine as it is strictly confined to this function. Another way would be to introduce an arch specific `lx_emul` interface to initialize `task->stack` but I am not sure that this is worth it.

jschlatow, 2022-07-28:
> @jschlatow thanks for the fixup. From my point of view your solution is fine as it is strictly confined to this function. Another way would be to introduce an arch specific `lx_emul` interface to initialize `task->stack` but I am not sure that this is worth it.

That was my thinking as well.

commit 9d383037 (2022-08-10): lx_emul: set task 'stack' member

Instead of allocating the stack via 'kmalloc' point it to the actual
stack used by the task. This addresses issues with the USB
host-controller driver where sometimes hub port enumeration is not
working.

Fixes #4522.
closed 2022-08-17 by chelmuth