genodelabs/genode #4525: gpu/intel: increase worst case CAP estimate

cnuke, 2022-06-09:
It may happen that the `Resource_guard` has only exactly *4* CAPs left and the check https://github.com/genodelabs/genode/blob/b6d313bbe64dd73b2c3d462bf19c3ebcba3af43a/repos/os/src/drivers/gpu/intel/main.cc#L1775 is false.

However, https://github.com/genodelabs/genode/blob/b6d313bbe64dd73b2c3d462bf19c3ebcba3af43a/repos/os/src/drivers/gpu/intel/main.cc#L1790 might consume 1 additional CAP and the `_resource_guard.withdraw` will therefor throw `Cap_quota::Limit_execeeded`.

cnuke, 2022-06-09:
Commit [58a0bfe](https://github.com/cnuke/genode/commit/58a0bfe47694994d05c34179fe2663d9483d4dfc) raises the estimate while commit [9989ad4](https://github.com/cnuke/genode/commit/9989ad4ee0168dddb9c2ea306115100cca9aaf7e) is up for debat how to deal with acounting mismatches. Since the liveliness of the GPU multiplexer is vital for e.g. Sculpt rather let the client abort. (Of course as noted in the comment that is a quick-fix and as is indicated by the commit's subject.)

nfeske, 2022-06-13:
> Since the liveliness of the GPU multiplexer is vital for e.g. Sculpt rather let the client abort.

Your suggestion makes sense to me. I merged your commit as e1620d407abaa5607d24baf81bcc21b7327008a7. @ssumpf do you have any reservation?

ssumpf, 2022-06-14:
> > Since the liveliness of the GPU multiplexer is vital for e.g. Sculpt rather let the client abort.
> 
> Your suggestion makes sense to me. I merged your commit as [e1620d4](https://github.com/genodelabs/genode/commit/e1620d407abaa5607d24baf81bcc21b7327008a7). @ssumpf do you have any reservation?

No objections from my side.

commit 1010267d (2022-08-10): gpu/intel: increase worst case CAP estimate

It may happen that 1 additional CAP is used during buffer allocation.

Fixes #4525.
closed 2022-08-17 by chelmuth