genodelabs/genode #4530: sculpt: allow to skip integration of depot archives

skalk, 2022-06-14:
When building a modular sculpt image, normally all packages referenced within a deploy configuration and within launcher configurations are build and integrated within the final image. Dependent on the scenario and the target platform the image to boot can exceed the maximum that is useable to boot.
To be able to nonetheless boot such rich scenarios an optional switch would be useful to skip the depot package integration into the boot image. Instead the packages have to be queried via network dynamically at runtime.

nfeske, 2022-06-16:
I like this suggestion but would prefer to avoid the use of an empty value.

How about allowing only the explicit values `omit` and `tar`? If the `DEPOT` environment variable does not exist, the default value `tar` would be assumed. If it exists but has a different value (e.g., should a user have a DEPOT environment variable with a different meaning), an error message would be nice.

The logic of evaluating the the environment variable could be implemented in a dedicated function `depot_integration` following the same pattern as the `sculpt_path` function. This function would return either `tar` or `omit`.

skalk, 2022-06-16:
@nfeske: I've tried to meet your suggestions with fixup commit d59395377afee4d97f84a2c9ca62734482a231a0

nfeske, 2022-06-17:
Thank you for addressing my remark. The name `depot_archive` is much nicer than my suggestion. I merged both commits to staging.

closed 2022-08-17 by chelmuth