genodelabs/genode #4534: nic_router slows down when there is no client in target domain

jschlatow, 2022-06-21:
I wrote a test component `nic_perf` in order to benchmark the throughput of the Nic and Uplink sessions. During my tests, I came across a weird performance degradation in the Nic router. The setup is as follows:

I have two nic_perf instances, one of which is sending UDP packets and the other receives these packets. The Nic router sits in between and performs routing between two domains. As soon as I remove the receiving nic_perf instance, the throughput drops drastically (1200Mbit/s vs 30Mbit/s). It can be reproduces on native hardware, in qemu as well as with base-linux. The corresponding test package is called `test-nic_perf_router` and available on my [nic_perf branch](https://github.com/jschlatow/genode/tree/nic_perf). The effect can be reproduced by removing the nic_perf_rx component from the package's runtime.

Enabling `verbose_packet_drop` and `verbose` provides the following hints:
```
[init -> depot_autopilot] 0.348 [init -> nic_router] [sender] free resources and retry to handle packet
[init -> depot_autopilot] 0.349 [init -> nic_router] [sender] using IP rule: 0.0.0.0/0
[init -> depot_autopilot] 0.349 [init -> nic_router] [sender] drop packet (insufficient resources)
```

@m-stein, any idea? For me it looks as if the router allocates an Arp_waiter for every packet.

m-stein, 2022-06-23:
I'm sorry that I have kept you waiting so long.

Your observation that the NIC router allocates an ARP waiter for each packet is correct and I can see that this can become a serious problem as you described it. I've had four ideas regarding this:
* A sessions packet handling is rate limited considering the number of packets handled, but the time for handling one packet may vary significantly from that of handling another.
* Packets of the same session with the same destination IP could wait for ARP with only one ARP waiter, reducing the amount of allocated memory per packet. However, a list element would still have to be allocated per packet.
* There could be a limit of pending packets for one destination IP at one session assuming that this limit should be hit only in exceptional cases (like the one you described). Further packets would be dropped without consuming time. This approach could be adapted to replace the waiting packets in a round robin fashion. Thereby, no further allocations would be needed and the packets waiting for ARP would always be the most recent ones.
* Presumably, there should be an idle timeout after which an ARP waiter would be destroyed. I'll have to read on this in the RFCs.

m-stein, 2024-06-18:
Issue https://github.com/genodelabs/genode/issues/4729 already added life timeouts for ARP waiters (default 10s). However, In the course of debugging failing nic_router_flood test, I found that the router sometimes drops packets because it aggregates to many ARP waiters. This is because the router might process up to 150 packets with one handler call. Given, that each of these packets has the same IP destination that is yet missing in the ARP cache of the domain, the router sends 150 ARP requests for the same IP address and creates 150 ARP waiters in order to remember all the blocking packets.

In an offline discussion @chelmuth and I concluded that we should, anyway, mitigate this effect by lowering the max number of packets in a batch and then also avoid re-sending ARP requests by keeping packets with the same IP destination in a list at a single ARP waiter. The latter means, that only the first attempt to communicate with an IP address triggers an ARP request and allocates an ARP waiter from the session quota, while all successive attempts merely allocate a list element in order to join the existing ARP waiter.

We refrained from going one step further, remove ARP waiters completely, and keep blocking packets directly at ARP cache entries, because this would allow for bogus ARP requests to fill up the ARP cache and thereby deny its service to sensible traffic.

chelmuth, 2024-06-18:
> We refrained from going one step further, remove ARP waiters completely, and keep blocking packets directly at ARP cache entries, because this would allow for bogus ARP requests to fill up the ARP cache and thereby deny its service to sensible traffic.

I have a follow-up question: How many entries does the current ARP support? As I understand the potential issue described above, it becomes more severe in large subnets, but is less critical in small /24 networks. Correct?

m-stein, 2024-06-19:
Currently, each domain has an ARP cache with 170 slots. In a /24 subnet, this would cover 2/3 of the possible addresses.

m-stein, 2024-06-19:
Whereas in a /16 network, it makes up for 0.25 % of the addresses.

m-stein, 2024-06-19:
Now that an ARP waiter can accumulate multiple packets that block for the ARP reply, imagine the following situation:

Peer P1 sends to peer P3 at time T1, thereby causing an ARP request and creating a waiter W with 10s lifetime. However, P3 is not alive yet. P3 becomes alive at T2 (> T1, < T1 + 10s) . Now, peer P2 sends to P3 at T3 (> T2, < T1 + 10s) and is added to W without sending an ARP request.

For P1 it was and is still a retry. However, in the past, P2 succeeded on the first try while now he also has to wait for up to 10s and retry. Is this fine?

m-stein, 2024-06-20:
@nfeske Thanks for review and merge! Let's see how the tests react to it.

commit e7f564cd (2024-06-20): nic_router: use C++20 function template syntax

Fix #4534
closed 2024-08-29 by chelmuth