genodelabs/genode #4536: nic_router packet throughput is diminished by exception handling

jschlatow, 2022-06-24:
Motivated by #4203, I started digging into network throughput benchmarking more systematically. My suspicion that the effect observed for falkon is just the pinnacle of an iceberg appears to be quite true. I have written a test component called ["nic_perf"](https://github.com/jschlatow/genode/tree/nic_perf) to start with benchmarking the packet stream interface and progressively add more components such as the nic_drv, nic_router, etc.

In order to get a base line, I benchmarked the throughput of the packet stream interface with two nic_perf components: One component transmitting a stream of UDP packets and the other receiving the same, without any packet inspection or copying. In this scenario, I achieved 13000Mbit/s on a t490s, 9000Mbit/s on an x260, and 1300Mbit/s on a Zynq board.

Adding the Nic router in between the two nic_perf components, dropped the throughput down to approx. 10% of the above results. I acquired some traces with the [trace_recorder](https://github.com/genodelabs/genode/issues/4352) of this scenario to look at the component interactions and the queue fill states. Looking at the traces, I could exclude that the throughput drop is due to eager signalling (i.e. missing batching of packet stream signals). Instead, the Nic router consumed most of the CPU time. I therefore instrumented the Nic router to find out where it spends its time. First, I was a bit baffled seeing that neither memcpy nor the checksum calculation was contributing significantly to the processing time of a packet. Instead, it turned out that most of the time is spent exception handling (i.e. between throw and catch). More precisely, the culprits in this benchmark are [interface.cc#L1231](https://github.com/genodelabs/genode/blob/master/repos/os/src/server/nic_router/interface.cc#L1231) and [interface.cc#L1273](https://github.com/genodelabs/genode/blob/master/repos/os/src/server/nic_router/interface.cc#L1273).

Out of curiosity, I simply commented out the related blocks and ran a test scenario in qemu. It increased the throughput from 75Mbit/s to 400Mbit/s!

_Note that this is not the skinniest bottleneck though: The missing batching of packet stream signals by the nic_drv appears to be the major culprit, yet this is a story for a separate issue._

jschlatow, 2022-06-24:
> Note that this is not the skinniest bottleneck though: The missing batching of packet stream signals by the nic_drv appears to be the major culprit, yet this is a story for a separate issue.

This was actually a false conclusion. As mentioned in #4538, the controller was operating only in 10Mbit mode. In consequence, the nic_drv handled the packets individually. When I make sure that the controller is in Gigabit mode, the driver is actually able to utilise the entire bandwith.

m-stein, 2022-06-27:
@jschlatow I very much appreciate your thorough investigation regarding net performance!

When I started writing the NIC router I was very much into the approach that code should just try out assumptions (e.g., a matching link state exists, a matching routing rule exists, a domain config is valid, etc.) in order of their priority and throw an exception in case the assumption doesn't hold. Today, however, I'm not so happy anymore with this idea. There is already a big note in my TODO list to get rid of at least most of the exception handling in the NIC router. Your observation reinforces this. The only thing that keeps me from solving this more general issue is that I currently have no time to do so.

Anyway, if it's fine with you, I try to fix the two places you pointed out.

jschlatow, 2022-06-29:
Fantastic! I just made a test run on an x250 where I got a 3x speedup with your fixes (600Mbit/s vs. 1800Mbit/s).

m-stein, 2022-06-29:
@jschlatow I'm happy that the commits have such a positive effect on your scenario :) Whenever I have some spare time in the near future, I'll remove further exception handling from the router.

nfeske, 2022-08-17:
Merged in master.

closed 2022-08-17 by nfeske