genodelabs/genode #4538: ipxe_nic_drv throughput drops to 10Mbit/s

jschlatow, 2022-06-24:


trimpim, 2022-06-27:
I have noticed this also. On Friday I tried to fix the issue (in the ipxr driver only) but wasn't successful.

With your explanation now its clear to we, why I couldn't succeed that way (I forgot to take the `platform_drv` in to account).

On the `imx8` platform driver one could add an other device that also uses the `PHY` clock and as long as one of the drivers still is running, the clock won't be turned off. I'm not sure if this is also possible with the `platform_drv` for the PC platform.

I will try this and write back with my findings.

chelmuth, 2023-06-20:
With the new pc_nic_drv merged in #4921, we will no longer pursue this issue of ipxe_nic_drv.

closed 2023-06-20 by chelmuth