genodelabs/genode #4542: dde_linux: improvements during pending i2c_hid driver port

chelmuth, 2022-06-29:
- fully enable printk format specifiers
- add devm_ioremap_uc()
- add lx_emul_backtrace()
- re-enable -Wformat

chelmuth, 2022-07-08:
@skalk and other dde_linux devs: could these changes be merged as is?

skalk, 2022-07-18:
@chelmuth: sorry for having overseen your request. I'm fully OK with those 4 commits.

cnuke, 2022-07-18:
In case of i.MX8 using [353a3fd](https://github.com/genodelabs/genode/commit/353a3fdac472d4b4cb660d126fe2f211764fc896) `vsnprintf()` triggers the execution of generated dummies, in particular `sprint_symbol_no_offset()`. This due to `CONFIG_KALLSYMS` being set where `symbol_string()` behaves differently.

(Implementing the functions as *NOP* in the common dummies compilation unit works but leads to information loss.)

closed 2022-08-17 by chelmuth