genodelabs/genode #4545: legacy_platform_drv: ACPI devices in configuration

chelmuth, 2022-06-29:
Currently, resource information for non-PCI devices is not available at runeimte. This patch enables the specification of statically known ACPI device resources on startup like follows.

```xml
<config>
  <policy label_prefix="driver">
    <device name="ACPI0000"/>
  </policy>
  <acpi-device hid="ACPI0000">
    <irq    number="99"/>
    <iomem  base="0xfc000000" size="0x100000"/>
    <ioport base="0x4000"     size="4"/>
  </acpi-device>
</config>
```

This is a preparatory commit for the pending i2c_hid driver.

chelmuth, 2022-07-08:
@alex-ab could you please review a252906718cc10033adf9fe97f4a3804837032a1 as it touches several places in the legacy platform driver

alex-ab, 2022-07-11:
I am fine with the commit, beside a naming remark regarding an member object.

commit 0a82c89f (2022-07-11): legacy_platform_drv: ACPI devices in configuration

Known ACPI device resources can be statically configured on startup like
follows.

  <config>
    <policy label_prefix="driver">
      <device name="ACPI0000"/>
    </policy>
    <device name="ACPI0000" typee="acpi">
      <irq number="99" mode="level" polarity="low"/>
      <io_mem        address="0xfc000000" size="0x100000"/>
      <io_port_range address="0x4000"     size="4"/>
    </device>
  </config>

Fixes #4545
closed 2022-08-17 by chelmuth