genodelabs/genode #4551: lxip occasionally slows down in high-throughput scenarios after some time

jschlatow, 2022-06-30:
I was benchmarking vfs_lxip with the nic_perf component on Sculpt. When transmitting an infinite stream of UDP packets via lxip, the packet throughput may drop significantly or even stop completely after some time. I started investigating and found out that lxip calls `schedule_timeout()` when it runs out of memory for skbs and never recovers from this state. I believe it fails to free the corresponding resources correctly or messes up the accounting. A similar issue can be triggered with the TCP_MAERTS test from netperf.

jschlatow, 2022-06-30:
I'm not sure how relevant this issue is for practical scenarios. Given that lxip needs a rewrite based on the new dde_linux approach anyway, I would rather not spend much time on this at the moment. Though, I added a warning by which we can identify that the aforementioned conditions have been triggered (see commit a2dc4b7).

nfeske, 2022-08-17:
@jschlatow how should we go about this issue?

jschlatow, 2022-08-17:
It's fine by me to close this issue and potentially reopen it if it becomes relevant in practical scenarios.

nfeske, 2022-08-17:
Thanks for the quick response.

closed 2022-08-17 by nfeske

Time0o, 2023-05-10:
After the introduction of https://github.com/genodelabs/genode/commit/a2dc4b7f07c5ffa9eb7e13afac45703d817a517f I have started seeing this error in a component that uses libcurl to upload file to some server via HTTP. As far as I can tell the upload still succeeds but I'm a bit concerned about the warning anyways.