genodelabs/genode #4556: pc_usb_host_drv: xhci host controller dies when heavily de/attaching USB3 hub

skalk, 2022-07-05:
When permanently and fastly detaching and attaching again (<1sec) a USB 3 hub at a Fujitsu Lifebook S983, I can observe a setup device command timeout, followed by a xhci_handshake timeout when the driver tries to shoot down the command ring buffer. It ends up with an unrecoverable xhci host controller dying:
```
[init -> usb_drv -> usb_drv] xhci_hcd 0000:00:01.0: Abort failed to stop command ring: -110
[init -> usb_drv -> usb_drv] xhci_hcd 0000:00:01.0: xHCI host controller not responding, assume dead
[init -> usb_drv -> usb_drv] xhci_hcd 0000:00:01.0: HC died; cleaning up
[init -> usb_drv -> usb_drv] xhci_hcd 0000:00:01.0: Timeout while waiting for setup device command
[init -> usb_drv -> usb_drv] usb 1-3: USB disconnect, device number 3
[init -> usb_drv -> usb_drv] usb 1-5: USB disconnect, device number 4
[init -> usb_drv -> usb_drv] usb 1-7: USB disconnect, device number 5
[init -> usb_drv -> usb_drv] usb 1-8: USB disconnect, device number 6
[init -> usb_drv -> usb_drv] usb 1-9: USB disconnect, device number 7
[init -> usb_drv -> usb_drv] usb 1-10: USB disconnect, device number 8
[init -> usb_drv -> usb_drv] Error: Function kobject_uevent_env not implemented yet!
[init -> usb_drv -> usb_drv] Backtrace follows:
[init -> usb_drv -> usb_drv] 0x106a6bb
[init -> usb_drv -> usb_drv] 0x109faf1
[init -> usb_drv -> usb_drv] 0x109ff47
[init -> usb_drv -> usb_drv] 0x108b766
[init -> usb_drv -> usb_drv] 0x10d4b34
[init -> usb_drv -> usb_drv] Will sleep forever...
```

ssumpf, 2022-07-06:
As a follow up. It seems like this regression has been caused by this https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb/host/xhci-ring.c?h=v5.15.52&id=5255660b208aebfdb71d574f3952cf48392f4306 in 5.14-rc2 and fixed by https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb/host/xhci-ring.c?h=v5.15.52&id=fa75f593c867cde772e2d87ca36ab735c4aa4a09 in 5.16-rc2 and back ported to 5.15. So our driver base version does not include the fix.

chelmuth, 2022-07-19:
I mark this issue fixed by 1db4792adb85d08a3eeb7d334416e74f2ae0423f.

commit 1db4792a (2022-07-15): pc/linux: add contrib patches

- intel_fb: avoid pagefault in fliphandler
- usb_host: avoid hang of usb xhci hardware

Fixes #4556
closed 2022-08-17 by chelmuth