genodelabs/genode #4559: Enable Mesa3D lima driver

cnuke, 2022-07-12:


cnuke, 2022-07-12:
Commit [1d398ba](https://github.com/cnuke/genode/commit/1d398bafe07de29714567912ae81300f51e40dc1) extends the *Gpu* session so that the lima driver is able to query the GPU virtual address, commit [fee6048](https://github.com/cnuke/genode/commit/fee6048840005624bf07fccfc100a56f91a9f336) provides the driver specific information and [9dd596b](https://github.com/cnuke/genode/commit/9dd596b02db5f05f18e2cb608ac713a2dc60061c) adds the driver port and libdrm back end.

cnuke, 2022-07-15:
I am currently working on a solution that fixes the archive issue introduced by the lima driver.

chelmuth, 2022-07-15:
Great, would you please adapt black_hole to your GPU-session changes too?

cnuke, 2022-07-18:
@chelmuth please merge the commits starting from [2dba2f7](https://github.com/cnuke/genode/commit/2dba2f787d22653400b270aba30a0ad1d24b50ae) to staging.

nfeske, 2022-08-17:
Merged to master.

commit eef9ea28 (2022-07-14): libports: add mesa/lima driver

This commit introduces the Mesa3D lima driver for Mali 400 series
GPUs.

Fixes #4559.
closed 2022-08-17 by nfeske