genodelabs/genode #4562: lx_emul: common emulation of softirq/tasklets insufficient

skalk, 2022-07-15:
The current emulation of the softirq/tasklet mechanisms within dde_linux are not sufficient to meet the requirements of several use-cases. It might lead to double-locks, wrong timing etc. Instead of trying to enrich the implementation even more, it is more pragmatic to use the original contribution sources here, like other drivers (e.g. imx usb host driver) already did.

skalk, 2022-07-19:
@chelmuth: I've compiled all drivers on x86 32/64bit and tested the USB driver on Qemu. However, several additional changes I had to apply because of the recent changes regarding kernel-time, I could not test on hardware using wifi and intel framebuffer. Nevertheless, I think it is acceptable to do a smoke-test on staging by applying the three commit above. Alternatively, I can test them on hardware tommorrow.

The only thing left to fix and close this issue is the wireguard port,which still uses the shadow variant of softirq.c.

skalk, 2022-07-19:
@chelmuth I had to add fixup commit 32360a9cb09f2a563f77740d1424f6cc2046a81f in addition to the previous three commits to fulfill missing dependencies for package creation.

chelmuth, 2022-07-19:
I merged your entire branch to staging. Let's cross fingers.

skalk, 2022-07-21:
@alex-ab: commit 58712e5 on my merge_to_staging branch enables SMP and NO_HZ_IDLE for virt_linux (and therefore wireguard), switches wireguard to use the original softirq implementation from the Linux sources, and removes our erroneous softirq emulation. Feel free to mege it to staging.

alex-ab, 2022-07-22:
> @alex-ab: commit [58712e5](https://github.com/genodelabs/genode/commit/58712e5fa06f95986a28e93ea2065e062f3920fa) on my merge_to_staging branch enables SMP and NO_HZ_IDLE for virt_linux (and [...]

Added to staging.

skalk, 2022-08-30:
This issue is fixed on master now.

closed 2022-08-30 by skalk