genodelabs/genode #4563: nic_router: don't send 'destination unreachable' on multicast packets

rolandbaer, 2022-07-16:
When the NIC router doesn't find a rule to handle a packet it sends an ICMP packet 'destination unreachable'. This is also done for multicast packets, but "they should be quietly discarded by hosts that do not support IP multicasting." (see RFC 1112, 3. LEVELS OF CONFORMANCE, Level 0: no support for IP multicasting.)

chelmuth, 2022-08-09:
Issue could be closed by #4564.

m-stein, 2022-08-16:
@rolandbaer @chelmuth I'm sorry for the delay! I reworked the original with a squash commit. Are you fine with that?
f7c180c8ce squash "nic_router: multicast packet handling"
0245f71963 nic_router: multicast packet handling

There are other conditions we should check according to the RFC but I left them out for now in order to make it easier for the commit to enter the upcoming Genode release and save me some time for other work. However, I've added an according FIXME note to the code.

rolandbaer, 2022-08-16:
@m-stein @chelmuth I'm fine with the squash.

Thinking about other conditions we should also think about different szenarios of the nic-router. The first is the behaviour of a router that sends information to other (real) networks, another is working as a dispatcher inside the host. For the second there should be, as far as know (no rfc to link to), no destination unreachable packets sent out. Maybe there could be some kind of configurable mode to choose between the different behoviours?
Do you have already an issue for the FIXME where we could discuss this further?

chelmuth, 2022-08-17:
Merged to staging.

8f228e3035b nic_router: no ICMP on unroutable IPv4 multicast

m-stein, 2022-08-27:
@rolandbaer I created issue #4594 for the remaining conditions and also thought that it would be sensible to make ICMP error functionality configurable.

commit 8f228e30 (2022-08-17): nic_router: no ICMP on unroutable IPv4 multicast

The NIC router used to send an ICMP "Destination Unreachable" packet as
response to every unroutable IPv4 packet. However, RFC 1812 section 4.3.2.7
defines certain properties that must be fullfilled by an incoming packet in
order to be answered with this type of ICMP. One requirement is that the packet
is no IPv4 multicast.

This commit prevents sending the mentioned ICMP response for unroutable IPv4
multicasts and instead drops them silently.

Fixes #4563
closed 2022-08-31 by chelmuth