genodelabs/genode #4566: Noticable lag on USB unplug in sessions to unrelated USB devices

chelmuth, 2022-07-19:
According to user report, this can be tested by moving a USB mouse while unplugging another USB device (e.g., thumbdrive). The mouse pointer hangs noticeably for a subsecond period. It was also reported that Isochronous audio streams experience stutter in this situation. The effect can be mitigated (but not eliminated) by removing `usb_reset_device()` on release. This is a regression with the new USB host driver.

chelmuth, 2022-07-19:
Based on my experiences in #4565 I suggest to remove `usb_reset_device()` and rely on the *unbind* magic in `usb_driver_release_interface()`. Further investigations of potential remaining lags after this change are pending.

skalk, 2022-07-22:
@alex-ab please merge commit c51bb8f to staging to partly fix this issue.

alex-ab, 2022-07-22:
> @alex-ab please merge commit [c51bb8f](https://github.com/genodelabs/genode/commit/c51bb8fba668ef45732daade9d7d36f82c44b555) to staging to partly fix this issue.

Added to staging.

chelmuth, 2022-08-10:
Users still report small but noticeable lag with d9407c74609864397464f7c8ed318bce3d9e3a99 applied.

skalk, 2022-08-30:
IMO we can only address this issue when changing the USB session in a way that almost everything is done asynchronously, especially opening/closing resp. claiming/releasing interfaces. Thereby, we also have to use an asynchronous way in closing USB sessions.

chelmuth, 2024-04-12:
@skalk may we close this issue with the USB renewal merged?

skalk, 2024-04-15:
Yes

closed 2024-04-15 by skalk