genodelabs/genode #4570: Don't omit frame pointer per default (proposal)

chelmuth, 2022-07-20:
According to discussions [[1](https://stackoverflow.com/questions/14666665/trying-to-understand-gcc-option-fomit-frame-pointer)] [[2](https://fedoraproject.org/wiki/Changes/fno-omit-frame-pointer)] [[3](https://stackoverflow.com/questions/13006371/does-omitting-the-frame-pointers-really-have-a-positive-effect-on-performance-an)] [[4](https://www.phoronix.com/scan.php?page=article&item=fedora-frame-pointer)], the per-default omission of the frame pointer in current GCC with `-O2` is questionable because of a slight performance impact only and undesired influence on profiling and debugging.

Additionally in the context of Genode, it often causes confusion because the frame-pointer-based backtrace (e.g., in `lx_emul_trace_and_stop()`) mostly ends up page-faulting. In these cases, users/developers repeatedly report pagefault errors despite the real error can be found some lines above in the log. From my experience, it's tiresome and also does not last long to explain the reasons behind this behavior and suggest to add `-fno-omit-frame-pointer` to *etc/tools.conf* in addition to ask for more logged diagnostics to hunt the real issue.

Another unpleasant effect when developing/debugging is that depot archives do not use the build directory configuration and, therefore, must be replaced by local builds (either in the build directory or alternatively by patching the build system) in case backtraces are desired.

For these reasons I propose to set `-fno-omit-frame-pointer` in *globals.mk* for all binaries built in Genode.

m-stein, 2022-07-20:
I think it's a good idea considering this.
Regarding the stack backtraces, I also had times when I was misled by the page fault in the first moment :) an alternative would be to detect framepointer omission and print an error at this point. I imagine calling dummy functions and looking at the fp diff would do this. But I'm also fine with your approach.

nfeske, 2022-08-08:
My interpretation of @chelmuth's references (page 5 of the Phoronix article in particular) is not so optimistic. So I'd not jump on it today.

To avoid the confusing page faults when printing backtraces w/o frame pointers, I really like @m-stein's idea of printing a diagnostic message.

closed 2023-06-16 by nfeske