genodelabs/genode #4574: qemu-usb library: ignore data requests for invalid endpoints

skalk, 2022-07-25:
A user reported a problem within the qemu-usb library backend for Genode. If an endpoint gets invalid, but the guest OS still sends some USB requests, we do not ignore that request and return USB_RET_NAK as intended by the qemu code, but still forward the invalid request to the USB session. On the other side, because the status of the packet does not get marked by our backend in this error case, it is implicitly handled as reuseable memory, which can lead to a page-fault later when the request gets cleaned up again. The solution is simply to mark the packet appropriatedly, and to not forward it to the USB session API.

skalk, 2022-07-25:
@cnuke please review commit bc1b389 and signal @alex-ab if it is approved to be merged to staging, thanks in advance!

cnuke, 2022-07-25:
@skalk @alex-ab LGTM.

alex-ab, 2022-07-27:
Added to staging

closed 2022-08-17 by chelmuth