genodelabs/genode #4575: dde_linux: remove exited tasks

cnuke, 2022-07-25:


cnuke, 2022-07-25:
Commits [de3e383](https://github.com/cnuke/genode/commit/de3e383ee9e40fcc314900365f3edbeaecbb44a6) and [85aea6e](https://github.com/cnuke/genode/commit/85aea6e0eb733936d23c50c30ba69037c5614312) address this problem.

That being said we might still leak the corresponding `task_struct` object which is supposedly freed by `__put_task_struct()` and what is perhaps even more pressing: I am not sure why some kworker threads are stopped in the first place. So far that seems to be normal in the `pc_wifi_drv` component but I also encountered this behaviour in the `pc_usb_host_drv` component.

cnuke, 2022-07-26:
Commit [c629605](https://github.com/cnuke/genode/commit/c629605923ce20f9614f3ce6b80eff31f2e55491) addresses the `task_struct` object clean up. All dde_linux components have to be adapted as they probably all contain a `__put_task_struct()` generated dummy and AFAICT all that make use of *workqueue.c* potentially suffer from the destruction of `kworker` tasks due to `idle_worker_timeout()` handling.

cnuke, 2022-08-05:
@alex-ab please remove the commit [00664fe](https://github.com/genodelabs/genode/commit/00664fe9e3b754d955012feb53f16eeff0266f0e) for now as introduces problems most apparently on the *x86_32* test system.

nfeske, 2022-10-12:
Fixed in master.

alex-ab, 2025-04-23:
Issue still persists