genodelabs/genode #4578: platform_drv: use new API for PCI drivers, deprecate old one

skalk, 2022-07-26:
Deprecate the old x86-only platform API, and use the recently developed generic one for all PCI drivers in Genode.

nfeske, 2022-09-23:
What a huge amount of work! Thank you for this well-curated commit series. Each single commit speaks of the great care you've put into this work. I especially like how you took the opportunity to modernize many places you visited, like the style of run scripts.

After reviewing each commit at the surface level and taking a closer look at the changes related to the components I originally authored (e.g., related to Sculpt, PS/2), I merged the entire commit series to staging.

I would very much appreciate the authors of the various drivers to skim the commits related to their respective drivers. E.g., @ssumpf  for VESA and AHCI, @cnuke for dde_bsd and GPU, @chelmuth for dde_ipxe.

skalk, 2022-09-26:
@nfeske: thank you for your nice words! I've a follow-up commit 62e30e9 which adds the pci header files to the os api package. Therewith some nightly problems of DDE Linux drivers for various platforms shall vanish.

cnuke, 2022-09-27:
@skalk thanks for your work, I adapted the `pci_audio_drv` commit in [e7f2de5](https://github.com/cnuke/genode/commit/e7f2de5544f0d7992a05ad6ddb1f6ea35f337a9e) (removed the drivers that require I/O port access since they are not of much use anyway) and tested the driver on the *Fuji4*.

(Since DMA snooping is only enabled for Intel devices, I'll check with non Intel systems.)

ssumpf, 2022-09-27:
@nfeske: ff26815 fixes `intel_fb` for Gen8+ Intel GPUs (tested on Macho, Fuji2).

cnuke, 2022-09-27:
Commit [2662d3a](https://github.com/cnuke/genode/commit/2662d3a40654e49f6ee78f334e5948584b760a7d) address the `nvme.run` issue with *seL4*.

skalk, 2022-09-27:
@chelmuth: no it does not. I think pci_decode would be the better place indeed.

ssumpf, 2022-09-27:
@chelmuth, @nfeske: e5ee664 is a suggestion that fixes AHCI on hardware. AHCI I/O memory is in BAR5. Hardware has multiple I/O memory regions, so we need to find the correct index.

cnuke, 2022-09-28:
Commit [0920c01](https://github.com/cnuke/genode/commit/0920c01ad27c78ef349655c34d1847c720ee3c64) takes care of the half-baked `dde_bsd` adjustments, sorry for the noise.

cnuke, 2022-09-28:
The `dde_bsd` driver currently does not work with systems featuring mutliple HDAudio devices where one of the exposed ones is not supported, e.g.
```
[init -> audio_drv] <devices>
[init -> audio_drv]     <device name="00:03.0" type="pci" used="true">
[init -> audio_drv]             <io_mem pci_bar="0"/>
[init -> audio_drv]             <irq/>
[init -> audio_drv]             <pci-config vendor_id="0x8086" device_id="0x160c" class="0x40300" revision="0x8" sub_vendor_id="0x1297" sub_device_id="0x4028"/>
[init -> audio_drv]     </device>
[init -> audio_drv]     <device name="00:1b.0" type="pci" used="false">
[init -> audio_drv]             <io_mem pci_bar="0"/>
[init -> audio_drv]             <irq/>
[init -> audio_drv]             <pci-config vendor_id="0x8086" device_id="0x9ca0" class="0x40300" revision="0x3" sub_vendor_id="0x1297" sub_device_id="0x4028"/>
[init -> audio_drv]     </device>
[init -> audio_drv] </devices>
```

as it will apparently always operate on the first one even when skipped. Pointing the driver via
```
<policy label="audio_drv -> "> <pci vendor_id="0x8086" device_id="0x9ca0"/> </policy>
```
to the proper one works. I'll look into this tomorrow if nobody beats me to the punch.

alex-ab, 2022-09-29:
Please consider fcc4e4c2716087e76d4a98a15926e43881ea31ad on my merge_to_staging. Due to changed allocation scheme by pci_decode (compared to old platform driver), it became apparent that insufficient amount of MSIs is supported. (test machine has room for 8 MSIs, but up to 12 was allocated by pci_decode).

chelmuth, 2022-09-29:
Thanks @ssumpf and @alex-ab, merged the fix to staging.

cnuke, 2022-09-29:
Commit [01b7e9f](https://github.com/cnuke/genode/commit/01b7e9f7ce51f67a558c3440c53cef6a7401c2b0) addresses skipping of unsupported devices in `dde_bsd`. Successfully tested on *test-x250*.

chelmuth, 2022-10-05:
The following code in *pci_decode* falsely skips reporting of interrupts for MSI/MSI-X-only PCI devices that do not configure an interrupt PIN (value = 0). Such devices are not uncommon in recent mobile device, e.g., Framework with Gen12 board and Fujitsu U7411.

https://github.com/genodelabs/genode/blob/40c2eae054558d8c975ed245f8cdb3c148018fe8/repos/os/src/app/pci_decode/main.cc#L151-L153

My first take was the following, but I'm unsure about its working on kernels that are not MSI-capable.

``` diff
+++ b/repos/os/src/app/pci_decode/main.cc
@@ -149,7 +149,7 @@ void Main::parse_pci_function(Bdf             bdf,
 
 
                /* IRQ pins count from 1-4 (INTA-D), zero means no IRQ defined */
-               if (!irq_pin)
+               if (!irq_pin && !(msi_capable && (msi_x || msi)))
                        return;
 
                gen.node("irq", [&]
```

chelmuth, 2022-10-05:
*platform_drv* is missing support for MSI-X-only devices like the Qemu NVMe model. To fix this we need an implementation of `Msi_x_capability::enable()` according to `Msi_capability::enable()`.

nfeske, 2022-10-05:
Fixup 8c9dc5936852d7abf0783405d4c9c2947c334b37 restricts the shared-interrupt mechanism to legacy IRQs (not MSI, MSI-X) only. With this change, the wifi driver starts up successfully on the Framework laptop. Thanks @chelmuth and @skalk.

chelmuth, 2022-10-05:
> Please merge commit [0a09e4f](https://github.com/cnuke/genode/commit/0a09e4f5e2b89a81a192111c1c7c4f4dceebadfe) as it is needed for applying the headset quirk on the S938.

Merged to staging.

skalk, 2022-10-05:
commit 9ec3136 introduces a PC-specific platform driver, which extends the generic functionality with the formerly known reset feature. commit f9ff824 integrates this driver into sculpt.

nfeske, 2022-10-05:
Merged to staging. The reset feature works perfectly on the Framework laptop.

skalk, 2022-10-05:
commit ed61e40 implements MSI-x support within the generic platform driver to support devices such as NVMe models in Qemu that support MSI-x only.

nfeske, 2022-10-05:
I cherry-picked https://github.com/genodelabs/genode/commit/ed61e402b173838df1c260e37450308196bd5a1b to staging to see how the automated tests fare.

While giving it a try on the Framework laptop, however, I observed that the NVME driver does no longer work with this change. It seems to block infinitely (presumably not getting any interrupt), so part_block stalls. Reverting the commit revives the NVME driver.

chelmuth, 2022-10-06:
This may be related to my commit comment https://github.com/genodelabs/genode/commit/5765d419cb7d9db4a7371580f7e937d619309bdb#commitcomment-85904372.

> Note, I did not change the MSI vs. MSI-X priority, just documented the current state.

It may be worth a try to change the priorities and prefer MSI over MSI-X. @m-stein could you try a staging Sculpt on your T480 and check if the wifi_drv still works as expected?

skalk, 2022-10-06:
@chelmuth: I agree that would be a short-term solution, but when we integrate the MSI-x solution, it would be good to know it is working at least on some real hardware.

chelmuth, 2022-10-06:
> @chelmuth: I agree that would be a short-term solution, but when we integrate the MSI-x solution, it would be good to know it is working at least on some real hardware.

Sure, just wanted to recollect what we learned when enabling the wifi_drv on @jschlatow's T480s - for this device we had to disable MSI-X IIRC.

nfeske, 2022-10-06:
> I would like to investigate the problems on real hardware, if I can borrow this hardware from you on Friday.

Of course. As I have the Framework laptop with me, I can also do some testing today, if this can be of any help.

chelmuth, 2022-10-06:
> I would like to investigate the problems on real hardware, if I can borrow this hardware from you on Friday. Till now, I did not had hardware in my hands to test the MSI-x feature, but only Qemu.

It seems *usb_hid_raw* is also broken in Qemu. The XHCI uses `<irq type="msi-x" number="1"/>`. My Qemu is 6.1.0.
.

alex-ab, 2022-10-06:
> @nfeske I would like to investigate the problems on real hardware, if I can borrow this hardware from you on Friday. Till now, I did not had hardware in my hands to test the MSI-x feature, but only Qemu. However, I've prepared a fixup commit on my merge_to_staging branch, which fixes the 32-bit issues the feature caused this night in the test cycles.

The 64bit test machine has a NVMe PCI card which has MSI-X and was tested each night with the old platform driver.

chelmuth, 2022-10-06:
> It seems _usb_hid_raw_ is also broken in Qemu. The XHCI uses `<irq type="msi-x" number="1"/>`. My Qemu is 6.1.0.

If I enforce MSI in platform_drv, the test succeeds. While doing so, I noticed that platform_drv ignores the IRQ type reported by pci_decode and just applies its own preference for MSI-X.

skalk, 2022-10-06:
@chelmuth @alex-ab thanks for your hints, I'll investigate and rework the MSI-x support tomorrow at the office.

ssumpf, 2022-10-07:
> Merged to staging. The reset feature works perfectly on the Framework laptop.

The reset feature freezes my T470 laptop. It worked with the old platform driver.

chelmuth, 2022-10-07:
> The reset feature freezes my T470 laptop. It worked with the old platform driver.

Same on X260.

ssumpf, 2022-10-07:
@chelmuth: [0d035ae](https://github.com/ssumpf/genode/commit/0d035aed7c1c360843693b596cc8ca304b24f173) is the tested and working commit.

chelmuth, 2022-10-07:
> @chelmuth: [0d035ae](https://github.com/ssumpf/genode/commit/0d035aed7c1c360843693b596cc8ca304b24f173) is the tested and working commit.

Merged slightly adapted e55a38485c390b0e0af2886e47bb320002512d86

nfeske, 2022-10-10:
Thanks @alex-ab. It's really nice to see the acpica mechanics becoming simpler. Thank you adjusting the documentation as well.

chelmuth, 2022-10-10:
@alex-ab could the following branch also be removed from acpica?

https://github.com/genodelabs/genode/blob/79471f04fe4a9083d5ffdff2178a00aedd4a3503/repos/libports/src/app/acpica/os.cc#L419-L443

Maybe, `Acpica::generate_report()` is no longer used too?

cnuke, 2022-10-11:
Commit [9f21bfa](https://github.com/cnuke/genode/commit/9f21bfa180eb61601f16ffe94a24635a9f344e14) adds the new Timer session requirement to the `wifi.run`, `audio_out.run` and `audio_in.run` run script.

cnuke, 2022-10-11:
Commit [6e5effc](https://github.com/cnuke/genode/commit/6e5effc07b2fbcbe5a6d809792769927943d908e) adds a quirk for using GSIs for certain HD audio devices.

alex-ab, 2022-10-11:
> @alex-ab could the following branch also be removed from acpica?
> 
> https://github.com/genodelabs/genode/blob/79471f04fe4a9083d5ffdff2178a00aedd4a3503/repos/libports/src/app/acpica/os.cc#L419-L443
> 
> Maybe, `Acpica::generate_report()` is no longer used too?

Yes, commit 315fdfc87c57b7d46dbb22fbc74b7cc97aff5208 removes the code.

alex-ab, 2022-10-11:
> Yes, commit [315fdfc](https://github.com/genodelabs/genode/commit/315fdfc87c57b7d46dbb22fbc74b7cc97aff5208) removes the code.

and b7232e4c6a4292948c66de8225a8291522dbb3cd.

alex-ab, 2022-10-11:
2b998325a1627f177f54ca9dec05b6061b2f453f fixes the routing issue of nvme and intel_fb tonight

chelmuth, 2022-10-11:
Thanks, @cnuke @alex-ab . Merged your commits to staging.