genodelabs/genode #4582: PKCS#11 support

m-stein, 2022-08-16:
This issue aims for supporting access to USB smart card devices via the PKCS#11 interface using a port of the OpenSC suite.

m-stein, 2022-08-16:
The main subject to this issue is the GOA port of the OpenSC PKCS#11 tool that can be found at https://github.com/m-stein/goa_projects/tree/master/pkcs11_tool. In order to build the tool, the following GOA patches are required:

https://github.com/m-stein/goa/commit/b9432d2d1add4cca50bc0646b4dce23c31270d7e autoconf: improve appliance of verbosity settings
https://github.com/m-stein/goa/commit/f231a3bd0a9e34bfdaf063c2717a273f5db784d4 autoconf: define CPP and CXXCPP for configure
https://github.com/m-stein/goa/commit/9ea55a630648e31d398b8e8e98d5724df188ac0c quirks: support using pcsc-lite api
https://github.com/m-stein/goa/commit/7024fa9b71498798ed0bde4ad0794a9ecb6b77b3 autoconf: autoreconf if there's only configure.ac

The following commit series adds a port of the OpenSC PKCS#11 library that is required by the tool. It also adds a run script for testing the tool and Sculpt support:

f485799b7d ports: pkcs11_tool run script
3ca2d1091d libports: port of OpenSC PKCS#11 library
5a26124902 pcsc-lite: fix inc dir base in depot case
e45afca6e3 pcsc-lite: src and api recipe
adcbd6960a driver_manager: export USB smart card devices

nfeske, 2022-08-16:
Thank you @m-stein for giving Goa a try, and for the welcome improvements. Could you take the Goa-specific changes the issue tracker at https://github.com/nfeske/goa/issues ?

How do you think of hosting the ported library and run script at the Genode world repository? I'd like to keep the scope of the Genode main repository limited to software that is fundamental to the feature set of Sculpt. Should pcsc-lite become such an ingredient in the future, we can relocate it to the Genode main repository then.

m-stein, 2022-08-16:
This is the related GOA issue: https://github.com/nfeske/goa/issues/19
This is the related Genode World issue: https://github.com/genodelabs/genode-world/issues/300

These are the Genode commits:
345cf4e83c pcsc-lite: fix inc dir base in depot case
af8872af86 pcsc-lite: src and api recipe
d9b78afb12 driver_manager: export USB smart card devices

m-stein, 2022-08-17:
I've added a fixup according to a discussion with @nfeske in the GOA issue:
abc8c0daa1 fixup "pcsc-lite: src and api recipe"

chelmuth, 2022-08-17:
Merged to staging.

d0a33e34da3 pcsc-lite: fix inc dir base in depot case
3286d37c64a pcsc-lite: src and api recipe

m-stein, 2022-08-18:
There's another commit required that somehow slipped through my fingers initially:
985b886d29 vfs_libusb: fix missing dependency to libusb

chelmuth, 2022-08-19:
Merged f64d1bbc307111875a34bdb6ee9384029eca7b37 to staging.

m-stein, 2022-08-19:
@chelmuth Thank you ;)

commit d0a33e34 (2022-08-17): pcsc-lite: fix inc dir base in depot case

Fixes #4582
closed 2022-08-31 by chelmuth