genodelabs/genode #4589: qt5: update to version 5.15.2

cproc, 2022-08-23:
Qt 5.15.2 is the latest Qt version with QtWebEngine support on FreeBSD.

cproc, 2022-08-23:
Fixed by e4c8d69.

To use the updated version, the Qt host tools need to be updated as well with the `tool/tool_chain_qt5` script.

chelmuth, 2022-08-23:
Thanks, I merged the commits and built the tools.

closed 2022-08-31 by chelmuth

chelmuth, 2024-10-18:
@cproc Why does the port come with *genode/api/include/QtGui/5.15.2/QtGui/qpa_genode/qgenodeplatformwindow.h* and *qtbase/src/plugins/platforms/genode/qgenodeplatformwindow.h* and why do those files diverge?

cproc, 2024-10-18:
> @cproc Why does the port come with _genode/api/include/QtGui/5.15.2/QtGui/qpa_genode/qgenodeplatformwindow.h_ and _qtbase/src/plugins/platforms/genode/qgenodeplatformwindow.h_ and why do those files diverge?

The `api` one is used by `QGenodeViewWidget` and `qt_avplay`. If they diverge it just means that I forgot to update the copy.

commit 4fb0d99d (2022-08-23): qt5: update to version 5.15.2

Fixes #4589
commit 82a57a10 (2022-08-31): qt5: update to version 5.15.2

Fixes #4589