genodelabs/genode #4593: Missing pc_usb_host_drv and usb_hid_drv

SamuraiCrow, 2022-08-26:
I'm trying out the Nova microhypervisor with the line:

make run/demo KERNEL=nova BOARD=pc

and I get the error:

Error: incomplete or missing recipe (genodelabs/src/pc_usb_host_drv genodelabs/src/usb_hid_drv)

I'm following the Genode Foundations 21.05 pdf and tried the typing in the suggested help message from the error handler:

update depot: /home/samuraicrow/Documents/genode/tool/depot/create genodelabs/bin/x86_64/base-nova genodelabs/bin/x86_64/demo genodelabs/bin/x86_64/global_keys_handler genodelabs/bin/x86_64/init genodelabs/bin/x86_64/nit_focus genodelabs/bin/x86_64/nitpicker genodelabs/bin/x86_64/report_rom genodelabs/bin/x86_64/rom_filter genodelabs/pkg/x86_64/drivers_interactive-pc CROSS_DEV_PREFIX=/usr/local/genode/tool/21.05/bin/genode-x86- DEPOT_DIR=/home/samuraicrow/Documents/genode/depot UPDATE_VERSIONS=1 FORCE=1 REBUILD= -j8 

and it executes but doesn't help.  When trying again it yields the same error.

I'm trying to run this on Manjaro Linux x86_64 XFCE.  My hardware is an 8 year old gaming rig with a 3rd gen i7 CPU and 32 GB of DDR3.  It has UEFI firmware and the USB3 is working under Linux.

Are there any suggestions of how to get around this?

chelmuth, 2022-08-26:
It seems your build-directory configuration and instructions used do not fit the source-code version of Genode. Could you please try to execute the following commands additionally to the instructions at https://genode.org/documentation/genode-foundations/21.05/getting_started/Obtaining_the_source_code.html and retry running the demo on NOVA?

```
> cd <genode-dir>
> git checkout -b 21.05
```

The background is that Genode's Git master version evolved much since 21.05 and build/configuration requirements changed too. If my hint above works you, we will amend the instruction in Genode Foundations accordingly.

chelmuth, 2022-08-26:
Discussion moved to the [mailing list](https://lists.genode.org/pipermail/users/2022-August/008042.html).

closed 2022-08-26 by chelmuth