genodelabs/genode #4599: Support the build of shared libraries via `make lib/<libname>`

nfeske, 2022-09-01:
During the discussion of #3975, I proposed adding first-class support for building shared libraries by specifying `lib/<libname>` arguments at the build system's command line. This brings the following immediate benefits:

* We can remove all pseudo targets for shared libraries that aren't referenced by any target. Examples are VFS plugins that are not linked against any target but dynamically loaded at runtime.
* The `LIB=` special case can be removed from the build system. In hindsight, this feature - added in commit a15a86b02485bbde78163559552784dc42fb38ac - was not fully thought through. In particular, it is limited to only one library.
* Run scripts can explicitly trigger the build of shared libraries via the `build` command.
* In contrast to the use of pseudo targets such as src/lib/vfs/ttf/target.mk (built via `make lib/vfs/ttf`, resulting in the shared object `vfs_ttf.lib.so`), the argument for building a library always matches the name of the library (`make lib/vfs_ttf`).

Down the road, this change opens up further improvements. E.g., once it is in place, all run scripts that require shared libraries could explicitly list those shared libraries in the `build` command, which makes the build process more transparent. Once this practice is established, we could remove the implicit build of shared library dependencies when building a target. This would make the life of Qt application developers using the Genode build system much easier because it would take the build (and many long-winded dependency checks) of the Qt framework out of the loop.

nfeske, 2022-09-01:
The commits at my `make_libs` branches https://github.com/nfeske/genode/commits/make_libs (genode) and https://github.com/nfeske/genode-world/commits/make_libs (genode-world) implement this change.

jschlatow, 2022-09-07:
> @jschlatow do you approve [genodelabs/genode-zynq@947a122](https://github.com/genodelabs/genode-zynq/commit/947a1227d345e9db2242579afc33031754c50acc)?

Yes, perfect.

jschlatow, 2022-09-08:
@nfeske, I noticed that the trace_recorder polices were not build anymore since the target.mk files have been removed accidentally. A fix is provided by the commits 3ae787d and ce6015e.

commit 4c341916 (2022-09-07): squash 'build: support library builds via lib/<libname>' (rm compatibility to LIB=<libname>)

With this patch, the LIB=<libname> option is no longer supported.

Fixes #4599
closed 2022-09-21 by chelmuth