genodelabs/genode #4600: depot_deploy fails if provided with an empty config

jschlatow, 2022-09-02:
I was trying to use depot_deploy for the automated deployment of depot packages on a Zynq board. The config is controlled by a custom _update_manager_ component. On startup, it provides an empty config that causes depot_deploy to fail on an unhandled `Nonexistent_sub_node` exception. Looking at the depot_deploy implementation, I noticed that it does not handle the case of missing `<static>` or `<common_routes>` nodes.

jschlatow, 2022-09-02:
Fixed by 3bb14ff.

nfeske, 2022-09-05:
Thank you for this welcome robustness improvement. Are you open for a minor coding-style suggestion?

You added an if-else block as follows.

```
if (config.has_sub_node("static")) {
    Xml_node static_config = config.sub_node("static");
    static_config.with_raw_content([&] (char const *start, size_t length) {
        xml.append(start, length); });
} else
    warning("config lacks <static> node");
```

I think when using `with_sub_node` as follows, the code becomes easier to follow.

```
if (!config.has_sub_node("static"))
    warning("config lacks <static> node");

config.with_sub_node("static", [&] (Xml_node const &static_config) {
    static_config.with_raw_content([&] (char const *start, size_t length) {
        xml.append(start, length); }); });
```

jschlatow, 2022-09-05:
I was thinking about `with_sub_node` but did not like doubling the `has_sub_node` call. I agree with the readability aspect, though.

Do you think there would be any use for a `with_sub_node` variant that takes a second functor which is called in case the sub node does not exist?

nfeske, 2022-09-05:
> Do you think there would be any use for a with_sub_node variant that takes a second functor which is called in case the sub node does not exist?

Yes, that would actually be a much better interface for `with_sub_node`. The current one should probably be named `with_optional_sub_node` so that the intent of the programmer to ignore the non-existing-node case gets reinforced.

What do you think? If you like this idea, would you be interested in pursuing the API change?

nfeske, 2022-09-08:
Thank you very much, @jschlatow! I merged all your commits to the respective staging branches.

closed 2022-09-21 by chelmuth